OllisGit/OctoPrint-SpoolManager #1: [Poll] How many spools do you have?

OllisGit, 2019-11-15:
The following poll helps my for implementing future features.

[image]
[image]
[image]
[image]

Looxoor, 2020-10-12:
53

Sembazuru, 2022-07-04:
138, including samples from MakerBox. With a dozen or so more spools and several dozen samples still to add.