OllisGit/OctoPrint-SpoolManager #2: [Request] Filament Profiles

stantond, 2019-12-18:
This looks like a very promising project!

FilamentManager seems to do the basics, but one idea it as implemented well is it's Profiles. This lets you store attributes common to multiple spools.

For example, I recently bought 4 spools of eSUN PLA+ in different colours. They print with the same temperatures, come on identical plastic spools, cost the same (usually), and so on.

It would be helpful to have something similar in SpoolManager, with the ability to override these values per instance of a spool.

OllisGit, 2020-01-11:
Hi @stantond ,

nice idea! I will put this to my todo list. 
But I think this will not be included in release 1.

Well...unfortunately at the moment I had a lot to do with my other plugins (DisplayLayerProgress and PrintJobHistory)...I didn't expect this.

Hopefully I can complete the tasks for the two plugins till end of the month, so that I can continue with the SpoolManager.

Please be patient.
Olli

OllisGit, 2020-07-16:
Hi @stantond ,
in the first release candidate you can use two features to fulfill your requirements:

- Mark a Spool as a template. Everytime you add a new spool the template spool is used and pre filled the form
- Copy a existing spool

Please test and give me a feedback.

Thx, in advance
Olli

closed 2020-08-25 by stale[bot]