OllisGit/OctoPrint-SpoolManager #3: What happened to the plugin?

NovaViper, 2020-01-09:
Im trying to install the plugin on Octoprint but it says not found. In fact, I go look at the releases and see that it's completely empty; even the link that you said to use to install the plugin says "Not Found". Could you fix this please?

OllisGit, 2020-01-11:
Hi @NovaViper !

Well...unfortunately at the moment I had a lot to do with my other plugins (DisplayLayerProgress and PrintJobHistory)...I didn't expect this.

Hopefully I can complete the tasks for the two plugins till end of the month, so that I can continue with the SpoolManager.

Please be patient.
Olli

Btw. I removed the download URL from the startpage.

I-like-stuff, 2020-05-22:
This is a great time to start on this plugin considering filament manager has been labelled abandoned, loads of people will be wanting a new maintained option.

OllisGit, 2020-05-25:
Yes, you'r right!
My other plugin "PrintJobHistory" is almost done, implementation is 100% completed. 

Now I cleanup everything and create documentation for the "offical plugin review-process".
These are the last tasks and then I can invest my full power to the SpoolManager-Plugin

Wolfsblvt, 2020-07-06:
That sounds great! Any further information on this?
I have recently set up my printer with OctoPrint again and I would love to have a nice history and spool tracking form the beginning, now that I want to open new spools for printing.

FilamentManager not working does leave a big hole in the features that your print history provides.  
Are you continuing development here soon?

OllisGit, 2020-07-16:
Hi,

RC2 is out and it works with OP1.3 and 1.4!
If you have any issues or feature request, please open a new issue.

BR
Olli

closed 2020-07-16 by OllisGit