OllisGit/OctoPrint-SpoolManager #4: [Enhancement] Include Weight from FilaWeigher

OllisGit, 2020-01-27:
Instead of calculated used weight use the real measured value.
http://automatedhome.party/2019/12/14/the-filaweigher-a-standalone-wifi-web-based-weight-sensor-for-3d-printer-filament-for-less-than-7/

gandy92, 2020-06-04:
I like that idea!

It would indeed already be quite useful if the weight of the bare spool, or better the initial gross weight (bare spool+material) could be specified for each spool in addition to the nominal initial material weight: With the filament manager, I sometimes forget to choose the correct material in octoprint and then have to manually correct the remaining amount of material by weighing the spool and subtracting what I originally determined as the spool weight (initial gross weight minus the nominal material weight). If measuring the current gross weight could be automated with an electronic scale that would be perfect, but providing this data manually would already help. Exposing this through the REST API could open this functionality for a whole range of hardware.

Probably a different feature request, but maybe related after all: If one would equip the spool holder not only with a scale but also an NFC reader and put an empty tag on the spool, we would only need to be able to store an arbitrary ID with the spool data (in this case the tag-ID) and the possibility to select the current spool via the REST API to fully automate spool handling as far as the SpoolManager is concerned.

OllisGit, 2020-07-16:
Hi @gandy92 ,

fyi: I added your feature request about scanning a spool to my mid-term column of the project planning board:
https://github.com/OllisGit/OctoPrint-SpoolManager/projects/1

gandy92, 2020-07-17:
Hi @OllisGit , cool, thanks. Good to see this project is picking up pace, can't wait to try it out.

dmberrycom, 2020-08-01:
I think utilising the weight is great, I would consider that the method to retrieve the current weight is flexible, so it can support different ways to fetch or have data pushed. For example at the moment I just use a simple a script on the pi that is directly connects to a HX711 & LoadCell and shows me he weight, there is an old octoprint plugin (https://github.com/dieki-n/Octoprint-Filament-Scale) that gets the data via this method but seems to be not maintained anymore.

closed 2020-09-11 by stale[bot]