OllisGit/OctoPrint-SpoolManager #7: "Error with Template" on Spools tab

JohnSh3p4rd, 2020-07-15:
Hi,

I just installed your release candidate. The install worked fine, but when I try to access the Spools tab, I get the following error:
"There was an error with the template index.jinja2 at line number 147: SpoolManager_tab_dialogs.jinja2 "

[image]

Also I noticed that the currency symbol is not saved, if I set it to €, click save, and reload the page, the text box is empty again. 

Do I need to install other plugins or dependencies? I am running OctoPrint 1.4.1rc2 with Python 2.7.16 on OctoPi 0.17.0.

If you need more information, I am happy to help, this is just my first bug report on GitHub, so there maybe information missing.

d4rken, 2020-07-16:
I'm on Octoprint `Version 1.4.0` + `V1.0.0rc1` and the currency symbol saves for me.

But I'm also getting:
> There was an error with the template index.jinja2 at line number 145: SpoolManager_tab_dialogs.jinja2

OllisGit, 2020-07-16:
Thanks for testing!
For some "mystical" reason my environment accept template-name with an uppercase letter.

Now I changed to the correct naming ```SpoolManager_tab_dialogs.jinja2``` instead of ```SpoolManager_tab_Dialogs.jinja2```

Please test the new RC2 and give me a feedback!

Thx, in advance
Olli

d4rken, 2020-07-17:
> For some "mystical" reason my environment accept template-name with an uppercase letter.

Windows machine?

> Please test the new RC2 and give me a feedback!

The issue has been fixed :+1:

JohnSh3p4rd, 2020-07-17:
Can confirm, the issue is also fixed for me. Thanks!

For testing, I rolled back from the OctoPrint version 1.4.1rc2 to the stable release 1.4.0. After that, the currency symbol is saved permanently.

closed 2020-07-17 by OllisGit