OllisGit/OctoPrint-SpoolManager #10: Dropdown spool selector does not display orange

d4rken, 2020-07-17:
On 1.0.0-RC2 and Octoprint 1.4.0.

"Orange" displaying.

```
<select id="selectSpoolSidebar" data-bind="options: allSpoolsForSidebar,
                     optionsCaption: 'Select Spool...',
                     optionsText: buildSpoolNameForSidebar,
                     value: selectedSpoolForSidebar
                    ">
<option value="">Select Spool...</option>
<option value="">black - Prusament PETG - JetBlack 3</option>
<option value="">    - Prusament PETG - PrusaOrange 3</option>
<!--                     event: { change: onSelectSpoolItemForSidebar } ">-->
</select>
```

d4rken, 2020-07-17:
Possibly because I accidentally touched the slider and it isn't the "default orange" anymore?

d4rken, 2020-07-17:
> Possibly because I accidentally touched the slider and it isn't the "default orange" anymore?

Works for other colors though.

d4rken, 2020-07-17:
Other colors too

```html
<option value="">Select Spool...</option>
<option value="">black - Prusament PETG - JetBlack 3</option>
<option value="">    - Prusament PETG - PrusaOrange 3</option>
<option value="">white - TPU 95A - White 1</option>
<option value="">dimgrey - TPU 98A - Gray 1</option>
<option value="">    - Prusament PETG - GalaxyBlack 1</option>
<option value="">    - Prusament PLA - AzureBlue 1</option>
<option value="">black - PLA - Black 1</option>
<option value="">black - Prusament PETG - JetBlack 1</option>
<option value="">    - PLA - Orange 1</option>
<option value="">    - PLA - Orange 1</option>
<option value="">    - Prusament PETG - PrusaOrange 1</option>
<option value="">    - Prusament PETG - PrusaOrange 2</option>
<option value="">red - PLA - Red 1</option>
<option value="">white - PLA - White 1</option>
<option value="">black - Prusament PETG - JetBlack 2</option>
<!--                     event: { change: onSelectSpoolItemForSidebar } ">-->
```

OllisGit, 2020-07-21:
Hi @d4rken ,

thanks for testing.

I use https://github.com/bgrins/TinyColor for translating the color-code to display names. 
Unfortunately I used the wrong orange color code in the color-picker. This is fixed in the next release.

You mentioned that other predefined colors not translated correctly. I could not reproduced this. Only orange was wrong.

Currently I used 8 predefined colors. Do you think I should add more, if yes which one?
[image]
Btw. I don't like the current Spool-Selection-Box, because I want colored entries, but this is not possible with the basic html stuff and I could find a good lib for that.

Here is the translation-list of all available colors:
```
    aliceblue: "f0f8ff",
    antiquewhite: "faebd7",
    aqua: "0ff",
    aquamarine: "7fffd4",
    azure: "f0ffff",
    beige: "f5f5dc",
    bisque: "ffe4c4",
    black: "000",
    blanchedalmond: "ffebcd",
    blue: "00f",
    blueviolet: "8a2be2",
    brown: "a52a2a",
    burlywood: "deb887",
    burntsienna: "ea7e5d",
    cadetblue: "5f9ea0",
    chartreuse: "7fff00",
    chocolate: "d2691e",
    coral: "ff7f50",
    cornflowerblue: "6495ed",
    cornsilk: "fff8dc",
    crimson: "dc143c",
    cyan: "0ff",
    darkblue: "00008b",
    darkcyan: "008b8b",
    darkgoldenrod: "b8860b",
    darkgray: "a9a9a9",
    darkgreen: "006400",
    darkgrey: "a9a9a9",
    darkkhaki: "bdb76b",
    darkmagenta: "8b008b",
    darkolivegreen: "556b2f",
    darkorange: "ff8c00",
    darkorchid: "9932cc",
    darkred: "8b0000",
    darksalmon: "e9967a",
    darkseagreen: "8fbc8f",
    darkslateblue: "483d8b",
    darkslategray: "2f4f4f",
    darkslategrey: "2f4f4f",
    darkturquoise: "00ced1",
    darkviolet: "9400d3",
    deeppink: "ff1493",
    deepskyblue: "00bfff",
    dimgray: "696969",
    dimgrey: "696969",
    dodgerblue: "1e90ff",
    firebrick: "b22222",
    floralwhite: "fffaf0",
    forestgreen: "228b22",
    fuchsia: "f0f",
    gainsboro: "dcdcdc",
    ghostwhite: "f8f8ff",
    gold: "ffd700",
    goldenrod: "daa520",
    gray: "808080",
    green: "008000",
    greenyellow: "adff2f",
    grey: "808080",
    honeydew: "f0fff0",
    hotpink: "ff69b4",
    indianred: "cd5c5c",
    indigo: "4b0082",
    ivory: "fffff0",
    khaki: "f0e68c",
    lavender: "e6e6fa",
    lavenderblush: "fff0f5",
    lawngreen: "7cfc00",
    lemonchiffon: "fffacd",
    lightblue: "add8e6",
    lightcoral: "f08080",
    lightcyan: "e0ffff",
    lightgoldenrodyellow: "fafad2",
    lightgray: "d3d3d3",
    lightgreen: "90ee90",
    lightgrey: "d3d3d3",
    lightpink: "ffb6c1",
    lightsalmon: "ffa07a",
    lightseagreen: "20b2aa",
    lightskyblue: "87cefa",
    lightslategray: "789",
    lightslategrey: "789",
    lightsteelblue: "b0c4de",
    lightyellow: "ffffe0",
    lime: "0f0",
    limegreen: "32cd32",
    linen: "faf0e6",
    magenta: "f0f",
    maroon: "800000",
    mediumaquamarine: "66cdaa",
    mediumblue: "0000cd",
    mediumorchid: "ba55d3",
    mediumpurple: "9370db",
    mediumseagreen: "3cb371",
    mediumslateblue: "7b68ee",
    mediumspringgreen: "00fa9a",
    mediumturquoise: "48d1cc",
    mediumvioletred: "c71585",
    midnightblue: "191970",
    mintcream: "f5fffa",
    mistyrose: "ffe4e1",
    moccasin: "ffe4b5",
    navajowhite: "ffdead",
    navy: "000080",
    oldlace: "fdf5e6",
    olive: "808000",
    olivedrab: "6b8e23",
    orange: "ffa500",
    orangered: "ff4500",
    orchid: "da70d6",
    palegoldenrod: "eee8aa",
    palegreen: "98fb98",
    paleturquoise: "afeeee",
    palevioletred: "db7093",
    papayawhip: "ffefd5",
    peachpuff: "ffdab9",
    peru: "cd853f",
    pink: "ffc0cb",
    plum: "dda0dd",
    powderblue: "b0e0e6",
    purple: "800080",
    red: "f00",
    rosybrown: "bc8f8f",
    royalblue: "4169e1",
    saddlebrown: "8b4513",
    salmon: "fa8072",
    sandybrown: "f4a460",
    seagreen: "2e8b57",
    seashell: "fff5ee",
    sienna: "a0522d",
    silver: "c0c0c0",
    skyblue: "87ceeb",
    slateblue: "6a5acd",
    slategray: "708090",
    slategrey: "708090",
    snow: "fffafa",
    springgreen: "00ff7f",
    steelblue: "4682b4",
    tan: "d2b48c",
    teal: "008080",
    thistle: "d8bfd8",
    tomato: "ff6347",
    turquoise: "40e0d0",
    violet: "ee82ee",
    wheat: "f5deb3",
    white: "fff",
    whitesmoke: "f5f5f5",
    yellow: "ff0",
    yellowgreen: "9acd32"
```

d4rken, 2020-07-22:
I think it's not possible to cover all possible colors that filament could have, so maybe more color sliders is not the best approach.

We could remove the individual color sliders and replace it with an RGB color picker + text field to enter a color name.

I'd type `AzureBlue` into the text box and select something in the RGB picker that looks somewhat similar to AzureBlue. Most manufacturers don't publish the exact color anyways. The code picked in the RGB picker could later be used to display the color in the spool list, for now just to have that data for later. 

For now we could display it in the list as `TypedInColorName (#PickedColorCode)`.

> Btw. I don't like the current Spool-Selection-Box, because I want colored entries, but this is not possible with the basic html stuff and I could find a good lib for that.

Is it possible to colorcode the text? Or part of the text, i.e. only the `TypedInColorName (#PickedColorCode)` part? Like wrap the text line with something. Example

```html
AzureBlue Spool 1 - 870g (87%) - Prusament PLA - AzureBlue <font color="#FFFFFF">(#FFFFFF)</p>
```

:point_up: Not sure if that syntax exists, WebDev isn't my strong suit :frowning: .

OllisGit, 2020-07-25:
Hi @d4rken !
fyi, I changed the spool selection list
[image]
As suggested, I also added the option to enter a custom color name. It will be used as a tooltip.
[image]

This enhancement will be in the next release.

OllisGit, 2020-07-28:
Hi @d4rken,
it is implemented in RC4.

Please test and give me a feedback.

Thx, in advance
Olli

d4rken, 2020-07-29:
:orange_heart: 

[image]

closed 2020-07-29 by d4rken