OllisGit/OctoPrint-SpoolManager #11: Show remaining filament in the spool dropdown menu

d4rken, 2020-07-17:
It would be useful to see how much filament is left on a spool when selecting one in the dropdown menu. This would help me select the right spool.

[image]

OllisGit, 2020-07-21:
Hii @d4rken ,

good idea! Will be available in the next release.

OllisGit, 2020-07-22:
Hi @d4rken ,

it is implemented in RC3.

Please test and give feedback.

Thx, in advance
Olli

d4rken, 2020-07-22:
It shows usage, but the values are too large (>100%):

```html
<option value="">Select Spool...</option>
<option value=""> - Prusament PLA - AzureBlue 1 (398.4%)</option>
<option value=""> - Prusament PETG - GalaxyBlack 1 (452.3%)</option>
<option value="">black - Prusament PETG - JetBlack 3 (990.7%)</option>
<option value=""> - Prusament PETG - PrusaOrange 3 (797.2%)</option>
<option value="">black - Prusament PETG - JetBlack 2 (0.0%)</option>
<option value="">white - TPU 95A - White 1 (631.0%)</option>
<option value="">dimgrey - TPU 98A - Gray 1 (510.4%)</option>
<option value="">black - PLA - Black 1 (0.0%)</option>
<option value="">black - Prusament PETG - JetBlack 1 (0.0%)</option>
<option value=""> - PLA - Orange 1 (0.0%)</option>
<option value=""> - Prusament PETG - PrusaOrange 1 (0.0%)</option>
<option value=""> - Prusament PETG - PrusaOrange 2 (0.0%)</option>
<option value="">red - PLA - Red 1 (0.0%)</option>
<option value="">white - PLA - White 1 (0.0%)</option>
<!--                     event: { change: onSelectSpoolItemForSidebar } ">-->
```

d4rken, 2020-07-22:
It displays the weight in gram as %.

Think about it, we should either display just the gram, or gram+%, but not just the % as people use difference spool sizes (250g,750g,1kg,2kg) so knowing how many grams are left would be the most useful.

I guess changing the % to g would be the fix here?

OllisGit, 2020-07-28:
Hi @d4rken,
it is implemented in RC4.

Please test and give me a feedback.

Thx, in advance
Olli

d4rken, 2020-07-29:
:smile: Nice 

[image]
[image]

closed 2020-07-29 by d4rken