OllisGit/OctoPrint-SpoolManager #8: Screenshots vs RC2?

d4rken, 2020-07-17:
> https://github.com/OllisGit/OctoPrint-SpoolManager/blob/master/screenshots/editSpool-dialog.png

The screenshots show additional fields such as "QR Code" that are not visible in the current 1.0.0-RC2.

Are the screenshots from an internal build?

OllisGit, 2020-07-21:
Hi @d4rken ,

sorry for not updating the Screenshots.

The mentioned screenshot shows an internal proof of concept. I marked that screen, because in the future I will support this scanning-feature.

In the meantime, please look always in the feature list with the checkboxes in the readme.md

d4rken, 2020-07-22:
No worries, just curious.  Looking forward to future releases :+1:

closed 2020-07-22 by d4rken