OllisGit/OctoPrint-SpoolManager #9: Used filament % is not displayed

d4rken, 2020-07-17:
On 1.0.0-RC2 and Octoprint 1.4.0.

After adding a small it does not display the remaining % in the spool list.

[image]
[image]

OllisGit, 2020-07-22:
Hi @d4rken ,

it is implemented in RC3.

Please test and give feedback.

Thx, in advance
Olli

d4rken, 2020-07-22:
Works :+1: 

[image]

closed 2020-07-22 by d4rken