OllisGit/OctoPrint-SpoolManager #13: How do I add spools?

NovaViper, 2020-07-19:
Hey, I just installed the plugin but there seems to be no way of adding spools. I go into the spool tab and add spool information but it never seems to save at all. I'm using Octoprint 1.4.0

[octoprint.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/4943835/octoprint.log)

portcqb, 2020-07-19:
I also have this issue even with all info filled out the save button does nothing in rc2. I also tried uninstalling rc2 with the delete data option and installed rc1 to test but doing that rc1 has no database like some config from rc2 stays behind after uninstall.

OllisGit, 2020-07-21:
Hi @NovaViper ,

thanks for providing the log file. I can see that your server is starting several times in safe-mode, but I didn't see any error during startup or during save.

@NovaViper , @portcqb 
Please turn on the plugin logging and restart the server any try again saving a spool.
Please press Strg + F5 for reloading the page without a cache.

[image]

Also please look into the data folder for the database ```spoolmanager.db```. Path could be found in the Plugin-Settings.

NovaViper, 2020-07-21:
I'm not getting any sort of useful info from the debugg flags. When I hit save on the spool, I get this

```
2020-07-21 19:58:23,831 - octoprint.plugins.logging - INFO - Setting logger octoprint.plugins.SpoolManager.DatabaseManager level to 10
2020-07-21 19:58:23,834 - octoprint.plugins.logging - INFO - Setting logger octoprint.plugins.SpoolManager.DatabaseManager.SQL level to 10
2020-07-21 19:58:23,836 - octoprint.plugins.logging - INFO - Setting logger octoprint.plugins.SpoolManager level to 10

```

portcqb, 2020-07-21:
I added debugs for all of the spoolmanager portions this is what happens when I attempt to add a spool also the db is in the folder it specifies 
2020-07-21 19:08:32,272 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT COUNT(1) FROM (SELECT 1 FROM "spo_spoolmodel" AS "t1") AS "_wrapped"', [])
2020-07-21 19:08:32,281 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT "t1"."databaseId", "t1"."created", "t1"."isTemplate", "t1"."displayName", "t1"."vendor", "t1"."material", "t1"."density", "t1"."diameter", "t1"."color", "t1"."temperature", "t1"."totalWeight", "t1"."usedWeight", "t1"."usedLength", "t1"."code", "t1"."firstUse", "t1"."lastUse", "t1"."purchasedFrom", "t1"."purchasedOn", "t1"."cost", "t1"."costUnit", "t1"."labels", "t1"."noteText", "t1"."noteDeltaFormat", "t1"."noteHtml" FROM "spo_spoolmodel" AS "t1" ORDER BY "t1"."lastUse" DESC LIMIT ? OFFSET ?', [30, 0])
2020-07-21 19:08:32,286 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT DISTINCT "t1"."vendor" FROM "spo_spoolmodel" AS "t1"', [])
2020-07-21 19:08:32,290 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT DISTINCT "t1"."material" FROM "spo_spoolmodel" AS "t1"', [])
2020-07-21 19:08:32,293 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT DISTINCT "t1"."labels" FROM "spo_spoolmodel" AS "t1"', [])
2020-07-21 19:08:32,301 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT "t1"."databaseId", "t1"."created", "t1"."isTemplate", "t1"."displayName", "t1"."vendor", "t1"."material", "t1"."density", "t1"."diameter", "t1"."color", "t1"."temperature", "t1"."totalWeight", "t1"."usedWeight", "t1"."usedLength", "t1"."code", "t1"."firstUse", "t1"."lastUse", "t1"."purchasedFrom", "t1"."purchasedOn", "t1"."cost", "t1"."costUnit", "t1"."labels", "t1"."noteText", "t1"."noteDeltaFormat", "t1"."noteHtml" FROM "spo_spoolmodel" AS "t1" WHERE ("t1"."isTemplate" = ?)', [True])

OllisGit, 2020-07-22:
In RC3 I added more log informations, maybe it helps to analyse the issue.

Also (as @portcqb ), please activate the sql-logging in the plugin-settings. After activation you need to restart your server.

Btw. if you use chrome, firefox please open the dev-tolls be pressing F12. Maybe there is some javascript error in the console.

portcqb, 2020-07-22:
OK I uninstalled RC2 and updated to RC3 and re-enabled the logging options when saving the info in log looks the same as before 
`2020-07-22 09:26:24,424 - octoprint.plugins.SpoolManager - DEBUG - API Load all spool
2020-07-22 09:26:24,429 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT COUNT(1) FROM (SELECT 1 FROM "spo_spoolmodel" AS "t1") AS "_wrapped"', [])
2020-07-22 09:26:24,441 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT "t1"."databaseId", "t1"."created", "t1"."isTemplate", "t1"."displayName", "t1"."vendor", "t1"."material", "t1"."density", "t1"."diameter", "t1"."color", "t1"."temperature", "t1"."totalWeight", "t1"."usedWeight", "t1"."usedLength", "t1"."code", "t1"."firstUse", "t1"."lastUse", "t1"."purchasedFrom", "t1"."purchasedOn", "t1"."cost", "t1"."costUnit", "t1"."labels", "t1"."noteText", "t1"."noteDeltaFormat", "t1"."noteHtml" FROM "spo_spoolmodel" AS "t1" ORDER BY "t1"."displayName" DESC LIMIT ? OFFSET ?', [10, 0])
2020-07-22 09:26:24,447 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT DISTINCT "t1"."vendor" FROM "spo_spoolmodel" AS "t1"', [])
2020-07-22 09:26:24,453 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT DISTINCT "t1"."material" FROM "spo_spoolmodel" AS "t1"', [])
2020-07-22 09:26:24,458 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT DISTINCT "t1"."labels" FROM "spo_spoolmodel" AS "t1"', [])
2020-07-22 09:26:24,465 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT "t1"."databaseId", "t1"."created", "t1"."isTemplate", "t1"."displayName", "t1"."vendor", "t1"."material", "t1"."density", "t1"."diameter", "t1"."color", "t1"."temperature", "t1"."totalWeight", "t1"."usedWeight", "t1"."usedLength", "t1"."code", "t1"."firstUse", "t1"."lastUse", "t1"."purchasedFrom", "t1"."purchasedOn", "t1"."cost", "t1"."costUnit", "t1"."labels", "t1"."noteText", "t1"."noteDeltaFormat", "t1"."noteHtml" FROM "spo_spoolmodel" AS "t1" WHERE ("t1"."isTemplate" = ?)', [True])`

Using F12 though an error in the chrome dev tool when save is pressed
`packed_plugins.js?e2080cb8:5096 Uncaught TypeError: self.noteEditor.getHtml is not a function
    at SpoolManagerViewModel.SpoolManagerEditSpoolDialog.saveSpoolItem (packed_plugins.js?e2080cb8:5096)
    at HTMLButtonElement.<anonymous> (packed_libs.js?3339d949:11395)
    at HTMLButtonElement.dispatch (packed_libs.js?3339d949:5184)
    at HTMLButtonElement.elemData.handle (packed_libs.js?3339d949:4992)
SpoolManagerEditSpoolDialog.saveSpoolItem @ packed_plugins.js?e2080cb8:5096
(anonymous) @ packed_libs.js?3339d949:11395
dispatch @ packed_libs.js?3339d949:5184
elemData.handle @ packed_libs.js?3339d949:4992`

OllisGit, 2020-07-28:
Thanks for the browser console-output.

This is a bug I didn't see, because in my system I also use the PrintJobHistory-Plugin that is also using the NoteEditor.

It is now fixed in RC4!

This issue is also tracked in #20, thats the reason why I am closing this issue.

closed 2020-07-28 by OllisGit