OllisGit/OctoPrint-SpoolManager #18: Magic Spool Selection Comment for API Submitted gcode Files

tduehr, 2020-07-25:
I submit most jobs through the API with Cura or Prusa Slic3r. Being able to select the spool in GCODE would help me track filament usage correctly without having to open a browser every time.

OllisGit, 2020-07-25:
Hi @tduehr ,

this is a good idea and this is already on my long-term list, see Backlog: https://github.com/OllisGit/OctoPrint-SpoolManager/projects/1

OllisGit, 2020-09-03:
Hi @tduehr ,
do you know a "standard" gcode-comand to identify a spool selection?
If I use a comment instead, like ```;spool_selection=My Best Spool``` then I need to parse the file at the beginning. But this is not possible for SD-Card prints, because OP has no access to this.
Or is there a "standard" comment in your slices for selecting a spool?

wchill, 2020-09-05:
@OllisGit maybe take advantage of the M118 gcode for SD card prints? So adding something like `M118 A1 spool:My Best Spool` to a gcode file before printing could signal the plugin to select that spool.

tduehr, 2020-09-07:
I don't use the SD card myself so I'm tempted to say it's not really something you can do. `M118` sounds like a good idea. It'd be nice if there was a way to do the spool selection that prusa slic3r uses.

How much of the print information from an SD print is available anyway? Should this be handled by some sort of manual process to add/modify a print?

xunleii, 2020-09-28:
I think using the `M118` command can be great and could be easily managed by the `on_sentGCodeHook` hook (cf. https://github.com/OllisGit/OctoPrint-SpoolManager/blob/master/octoprint_SpoolManager/__init__.py#L367). But, I can't see how we can ignore the Spool warning message without reading the GCode before.

Also, I suggest `M118 A1 spool:<filament_id>:<spool_name>` (or something like that) instead of ``M118 A1 spool:<spool_name>` to prepare for the multi-filament feature.

What do you think ?

wchill, 2020-09-28:
Better would be to replicate what OctoPrint-ActionCommandsPlugin is doing:

https://github.com/benlye/OctoPrint-ActionCommandsPlugin/blob/ed03e15e96f52140b4066d58ba069e3612cca30d/octoprint_actioncommands/__init__.py#L24

The hook there lets you inspect serial input coming from the printer, which works well when you use M118.

wchill, 2020-09-29:
@OllisGit do you mind if I take this feature? I need something to work on for Hacktoberfest and I just recently got a MMU2S so it'll be useful for me anyway.

samwiseg0, 2021-03-06:
I was going to put in this request. Is there any chance we ran re open this issue since I think it would be a great idea since you can set filament gcode in prusaslicer and possibility others?

OllisGit, 2022-01-16:
fyi: Not closed any more, because I want to implement this in one of the next releases.

Fabietto996, 2023-01-14:
Are there any news about this addition?