OllisGit/OctoPrint-SpoolManager #19: Post Print Spool Selection

tduehr, 2020-07-25:
I often forget to change the spool when printing the same model multiple times. It would be nice if there was a way to correct the spool used for a print after completion.

OllisGit, 2020-07-25:
Hi @tduehr ,

please describe your request in more detail. 

What do you expect from the plugin, what values should be changed, how should a dialog look like, .....

Btw. you can activate the checkbox: "Reminder for verifying the selected spool." Then each time you start a print a popup will remind you to check the current selected spool.

d4rken, 2020-07-29:
I like the idea, I also often forget to select the correct spool.
I'm not sure how this could nicely integrated into the "print-flow" though.

Seems like something that could maybe integrated with PrintJobHistory?

tduehr, 2020-07-29:
I use that checkbox in FilamentManager currently. It doesn't help much when the print job is submitted through the API. Basically, I'd like to retroactively assign a spool to historic print jobs. Having this working would largely require the remaining spool to be determined live through the print job history, and may require fake/waste print jobs to be created for spool corrections (e.g., loss due to repairs or calibration routines).

tduehr, 2020-07-30:
As, @d4rken suggested, this may be best achieved by using PJH as a dependency and using the job history to update the remaining spool contents. This could even be done on an optional dependency basis with SpoolManager falling back to current behavior if PJH is not available.

OllisGit, 2020-09-03:
fyi: Closed tracked in PJH-Issue Tracker.

closed 2020-09-03 by OllisGit