OllisGit/OctoPrint-SpoolManager #20: self.noteEditor.getHtml is not a function

drawlerr, 2020-07-25:
```packed_core.js?3514a079:12054 
Uncaught TypeError: self.noteEditor.getHtml is not a function
    at SpoolManagerViewModel.SpoolManagerEditSpoolDialog.saveSpoolItem (packed_plugins.js?b5ef67ee:3935)
    at HTMLButtonElement.<anonymous> (packed_libs.js?3339d949:11395)
    at HTMLButtonElement.dispatch (packed_libs.js?3339d949:5184)
    at HTMLButtonElement.elemData.handle (packed_libs.js?3339d949:4992)
    at HTMLButtonElement.sentryWrapped (packed_core.js?3514a079:12054)```
Installed from master zip earlier today (2020-07-25)

OllisGit, 2020-07-28:
Hi @drawlerr ,
this is fixed in RC4!

Unfortunately I didn't see that bug in my environment, because my other Plugin "PrintJobHistory" use also the NoteEditor with the fixed behaviour.

Please test and give me a feedback.

drawlerr, 2020-08-01:
Thanks, @OllisGit!  Latest RC4 works for me.

closed 2020-08-01 by drawlerr