OllisGit/OctoPrint-SpoolManager #21: Default spring steel sheet for each spool

opTimos-A, 2020-07-27:
Hi,

would it be possible that a spool can be linked to a spring steel sheet for the headbed, that the Printer (in my case Prusa i3MK3s MMU2) select direcly the correct sheet. Actually I use two different ones PLA smooth one and PETG with the structured one. And i always forgot to change this setting in the Printer..... 
maybe in two steps. Short time development just ask befor printing and long time send correct GCODE to the printer.

OllisGit, 2020-07-28:
hmm..currently there is a PopUp Window to remind you to select the correct spool, just imagine that this is also a reminder to select the correct plate ;-)

Because I don't have such printer, what is the correct g-code for selecting the "sheet for the headbed"? 
Or do you mean selecting/switching the right spool during printing? This feature is already in my backlog and also requested in #18.

opTimos-A, 2020-07-28:
Hi @OllisGit,
Yes you are right, but than I have to go to the printer instead just click on the PC 😜
Thanks for your quick reply. I found a article that mentions M850. 

https://github.com/prusa3d/Prusa-Firmware/pull/2201

closed 2020-08-19 by OllisGit