OllisGit/OctoPrint-SpoolManager #22: Weight Calculations

primaxius, 2020-07-28:
You should integrate the densities right in with a material drop down with option to modify them. Use the weight to determine length remaining. i.e. you select PLA it sets the density to 1.24 but allows you to modify it.

Also you should separate the weights a bit ...

Spool Size (Weight you bought i.e. 1000g, 750g, 500g, etc...)
Spool Weight (Weight of the spool when empty)
Total Weight (on a new spool this should be i.e. 1000g+230g=1230g)
Remaining Weight (total weight-spool weight)

Length remaining would be a calculation based on https://bitfab.io/blog/3d-printing-materials-densities/

OllisGit, 2020-07-28:
Hi @primaxius ,

thanks for your suggestions!

I was also thinking about pre defining materials (was almost implemented), but then I decided to start with just a suggestion box from already used materials (without density).
Maybe I will implement this feature later (after V1)

To capture the weight of the spool also a good idea, because I know that some people reuse spools instead of buying filament with an included spool.
I will but this feature to my backlog.

Btw. your mentioned calculation link is already in the Edit-Dialog, see screenshot in the readme ;-)

OllisGit, 2020-08-20:
Hi @primaxius,
in the latest version 1.0.0 the material list is now prefilled and also after selection, the density field is pre filled with values from the mentioned web-page.
Please test and give feedback.

The separation of the weight as suggested is not implemented. 
It is now tracked in a separate issue #29

closed 2020-09-29 by stale[bot]