OllisGit/OctoPrint-SpoolManager #23: Export Legacy: Filament Manager Database used length rounding

RobeeeJay, 2020-07-28:
It's a brilliant idea to include an export that puts things in the right format from Filament Manager ❤️

I noticed though that the Used length (mm) column is a float but the import expects an integer. If I rounded it to an int it imported perfectly.

Another thing I noticed was after importing you had to refresh the page for the UI to show the new spools.

Thank you for this plugin! I will start testing it in more depth as I print things.

RobeeeJay, 2020-07-28:
Actually a related bug I just noticed, the Used Weight is blank... I can't see that being exported.

OllisGit, 2020-08-20:
Hi @RobeeeJay ,
is fixed in newest Version 1.0.0.

Please test and give feedback.

Thx, in advance
Olli

RobeeeJay, 2020-08-20:
It works! Thank you! I will run this alongside Filament Manager for the next week and see how it goes!

closed 2020-08-20 by RobeeeJay