OllisGit/OctoPrint-SpoolManager #24: Warning of insufficient material before print is started?

Spacemarine2018, 2020-08-02:
I'm running a small print farm and I want to avoid changing a spool in the middle of a print. Can SpoolManager give a warning, if the spool has insufficient material left?

Ideally, I'm just running one job after the other until SpoolManager gives a clear warning (or even blocks the START button) until I confirm that the spool is exchanged, so the system becomes idiot-proof.

I'm currently only printing one material, so a button "new spool has ben inserted" would be enough for normal operation.

OllisGit, 2020-08-20:
Hi @Spacemarine2018,
please test newest Version 1.0.0 and give feedback.

Thx, in advance
Olli

closed 2020-10-01 by stale[bot]

manolitto, 2020-11-15:
Is there also a chance to automatically pause or abort the print when started via API and there is not enough filament left?
I usually start my print jobs directly from SuperSlic3r with "autostart" option. Prusa Mini with current firmware is currently not able to pause when printing via OctoPrint and running out of material.
This would be a cool feature and would make it fool proof via API either.