OllisGit/OctoPrint-SpoolManager #26: Does it update the usage after a print?

spanzetta, 2020-08-07:
I installed it but I found that it doesn't update the used filament for a print
Is it something not implemented yet or do I have some problem?
The quantity of the filament remain always the same even after a print (done with THAT specific spool filament selected before starting the print)
Thanks

ScottEgan, 2020-08-12:
I am also having this issue. Will try to track down some logs for more info.

OllisGit, 2020-08-20:
hmm..I could not reproduce the issue, but in the newest version 1.0.0 I added more log-statements.
So, if you test again, please attach the octoprint.log to this issue.

Thx, in advance
Olli

ScottEgan, 2020-08-20:
I'll install 1.0.0 later today and test it again. Do you think it might have something to do with also having the FilamentManager installed? @spanzetta do you have FilamentManager installed too?

OllisGit, 2020-08-20:
There should be no conflict between both plugins.

ScottEgan, 2020-08-20:
Just did some testing. Looks like the usage is correctly updated after the print in the spools tab but not on the state panel (see attached picture). The state panel updates if I refresh the page.
[image]

OllisGit, 2020-08-20:
That shouldn't be the case! I will take al look on this.

skimaniac, 2020-08-25:
I have the same problem. to test properly I started a new reel and I printed 2 small projects, here is the info gathered by Print Job History

Job 1: 1.81m and 5.44g used
Job 2: 4.26m and 12.82g used
TOTAL: 6,07m and 18,26g used

SpoolsManager recorded -1130mm and -2.9g
FilamentManager recordes 21g used for the 2 job

skimaniac, 2020-08-27:
I think I found the problem, the octoprint log (octoprint.log) was huge, 350 meg, and I was not able to download it to verify it, finally I deleted it and since that the SpoolManager Plugin works as it should. :)

spanzetta, 2020-08-27:
> I'll install 1.0.0 later today and test it again. Do you think it might have something to do with also having the FilamentManager installed? @spanzetta do you have FilamentManager installed too?

Sorry for the delay but I am on vacation..
I will check and update when I come back home 
Thanks

5teveO, 2020-08-31:
HI Guys as I am not having the filament usage updated also - does this work also for SD card printed jobs (ie started from octoprint) and is it reliant on any other plugins? - I just want to eliminate my newness and any misunderstanding of how this works.

OllisGit, 2020-09-01:
Hi,

the new version 1.1.0 is out now. This version auto reload the values from the state panel.

Please test and give feedback.

@5teveO I never tested the sd-card approach and at the moment my printer is not working. But maybe someone else could answer the question.

Thx, in advance
Olli

5teveO, 2020-09-05:
@OllisGit thanks for the reply - I have updated to latest but no sd card reporting. I have stuttering issues with octoprint so am avoiding printing via the virtual sdcard currently - just starting jobs via octoprint from the printer SD card. No problem just have to remember to add the usage 👍

spanzetta, 2020-09-05:
now it seems to work fine!
Great job..
Thanks

RobeeeJay, 2020-09-05:
@5teveO I have been getting stuttering issues too, there is always a huge CPU spike when it happens, I wonder if it's a plugin but I'm not sure what debug/log options to set that might indicate which plugin it is. :(

OllisGit, 2020-09-07:
Hi @RobeeeJay, @5teveO  ,
if your are using DisplayLayerProgress, then start disabling this plugin. Because a lot of people reported different kind of issues. Maybe that fix your "stuttering".

closed 2020-09-07 by OllisGit

RobeeeJay, 2020-09-07:
Unfortunately no, I'm not running DLP :(

The only 3rd party ones are these:

- Access Anywhere - The Spaghetti Detective
- Bed Visualizer
- Cost Estimation
- PrintJobHistory
- PrusaSlicer Thumbnails
- Resource Monitor
- SpoolManager (obviously!)
- Tab Order
- Themeify

Maybe The Spaghetti Detective is a possible cause, but I think SpoolManager is the only other possible cause, but without knowing what to log and what level to log at I can't tell. :(