OllisGit/OctoPrint-SpoolManager #29: [Enhancement] Add additional Weights

OllisGit, 2020-08-20:
Spool Size (Weight you bought i.e. 1000g, 750g, 500g, etc...)
Spool Weight (Weight of the spool when empty)
Total Weight (on a new spool this should be i.e. 1000g+230g=1230g)
Remaining Weight (total weight-spool weight)

initial request: #22

hamster65, 2020-08-24:
This is a great plugin... I just added all of my spools.
Having net and gross weight (and automatic calculation of remaining material based on that) would be very handy. I wrote the gross  and the advertised net weight on every spool. That allows me to calculate or check the remaining material. I don't fully trust automatic odometers because it happens that you select the wrong spool, or loose data, or filament is not counted or....

OllisGit, 2020-09-03:
Added in latest version 1.2.0

Please test and give feedback

Thx, in advance
Olli

5teveO, 2020-09-10:
Im a little confused here... not unusual!

So I add my spools with gross weight and add the spool weight.  You calculate the remaining weight, which is fine for a sanity check , but what about the remaining filament weight? This should be Gross weight - Spool weight - usage... Or am I missing something?

Also the length - should this not be auto calculated? or... would it be an idea to have a button next to the length for calculate and it inputs the details based on weight (gross - spool) , diameter and density?

Just a thought as I have to go online to a filament calculator to calc the length (only because i'm lazy!)

Thanks for a great plugin too 👍 

Steve

ericsnis, 2020-09-10:
I'm not sure if this is expected behavior, but the empty spool weight doesn't seem to be subtracted from the total spool weight to calculate the remaining material. The feature itself is helpful so we can weigh an unknown spool and an identical unknown spool.
I was expecting behavior like what 5teveO was suggesting (Gross - empty spool - used weight = remaining weight).

SpoolManager Plugin 1.2.0
OctoPrint: 1.4.2

closed 2020-10-21 by stale[bot]