OllisGit/OctoPrint-SpoolManager #34: M600 support

blackboxgeeks, 2020-08-24:
Will you be adding in support for M600 gcode commands?

OllisGit, 2020-08-24:
Hi @blackboxgeeks ,

I added your feature request to the long term column: https://github.com/OllisGit/OctoPrint-SpoolManager/projects/1

NLH-Software, 2020-09-30:
Isn't this something like #18 
But would like to see this implemented, too.
Had a print two days ago, which used a filament change. Afterwards I had to edit the spools to correct the used lengths and weights manually.

codingcatgirl, 2021-05-14:
Hm. this shouldn't be closed. i'd really likeif i had *any* way to change the spool mid-print, doesn't even matter how complicated it is. With Filament-manager you can pause the print and change the filament selection. With SpoolManager, not even that is possible. This makes it impossible to come up with any workarounds for multi-filament prints until that feature is implemented. 

I feel like a lot of people (including me) are in dire need for (at least) a workaround of this issue,

OllisGit, 2022-01-16:
Hi,

I retested the behaviour with the latest release (V1.6.0) with this test-script in my virtual printer.
I changed the spool during the pause and at the end of the print both spools were updated correctly. 
```
G28
M83; relative extruder
G1 F1000
G0 E1000
M600
G0 E500
```

Please do me a favour and test this in your environment.

Thx, in advance
Olli

closed 2022-02-27 by github-actions[bot]