OllisGit/OctoPrint-SpoolManager #35: Flow Rate Compensation (feature request)

NameOfTheDragon, 2020-08-24:
Hi, this is a nice replacement for Filament Manager, good job.

One thing that I always felt was obviously missing from Filament Manager (and now in Spool Manager) was the ability to record a flow-rate compensation value for each spool, and then be able to use that to send a `M221 S<percent>` command in the pre-print GCODE. Flow rate is supposed to be calibrated per spool, and it is suggested to write this value on the spool label. However it would be even more useful to be able to enter this value into spools in the database and then be able to have OctoPrint automatically set the correct flow rate compensation percentage.

Does that make sense?

OllisGit, 2020-08-24:
Hi @NameOfTheDragon ,
I never heard about "Flow Rate Compensation". 

It sound easy to implement:
- New field in Database / UI to grab a number value (no float). E.g. 150%
- If value is present in the selected spool, just send the ```M221 S150``` as one of the first command after the print is started 

Did I missed something?

NameOfTheDragon, 2020-08-25:
Ah, my apologies, perhaps it is not as well known as I had assumed. Here's some background info:
https://teachingtechyt.github.io/calibration.html#flow
https://e3d-online.dozuki.com/Guide/Flow+rate+(Extrusion+multiplier)+calibration+guide./89

> ## Why do I need to change the flow rate? ##
> Different types filaments have different densities, and even different colors of the same filament brand can have different 
> extrusion properties which can lead to either under or over extrusion which creates defects in the print quality that you obtain. 
> Calibrating your flow rate will also help improve the accuracy of the printed geometry compared to the original geometry of the CAD model.

Some sources recommend that calibration should be performed for every new roll of filament, and the calculated percentage written on the spool for future reference. Personally, I do the calibration for each manufacturer, filament type and color and then if I get another roll of the same color I assume it will be the same (this could work well with the concept of "template spools")..

I'd say you understand the request correctly. However, rather than having the plugin inject G-Code, I wonder if it wouldn't be better to make the flow-rate percentage available as a variable that can then be used in OctoPrint's GCode Scripts (as documented at https://docs.octoprint.org/en/master/features/gcode_scripts.html#context). I don't know if this is actually possible, but it seems like it should be. Also note that the flow rate percentage will be different for each extruder, so any GCode would need to be sent any time there is a tool change.

OllisGit, 2020-09-03:
In the latest version 1.2.0 you can store the flow rate comp. value, but this is not reflected to the printer. I need more analysing/testing. 

Please test and give feedback

Thx, in advance
Olli

NameOfTheDragon, 2020-09-06:
I gave that a try and the value is there and is saved. The first thing I tried was entering "100.5" and was surprised when it was rounded down to "100". I checked the documentation and source code for both Marlin and Klipper. They both actually only accept an integer for `M221 S` so I guess that's OK. Looking goof so far. Now all I need is some way to use that value in a GCode script.

OllisGit, 2020-09-07:
Hi @NameOfTheDragon ,

what display resolution do you use?
If I reduce to a 1024 width, everything looks good. Maybe it is related to your custom theming.
[image]

NameOfTheDragon, 2020-09-08:
That's interesting. I'm using the Themify plugin with the "dark" theme selected and my display resolution is 1920x1080. Next time I have the printer powered up I'll try disabling that plugin to see if it has any effect.

[image]<https://user-images.githubusercontent.com/24205767/92411982-5e49f300-f14a-11ea-8def-8c5f9978084f.png>

OllisGit, 2021-04-15:
UI-Bug duplicated, see #112

closed 2021-04-15 by OllisGit