OllisGit/OctoPrint-SpoolManager #36: Cannot import from .csv

molodeztom, 2020-08-24:
I export existing database to .csv. Without editing I import the same .csv. I got an error:
Mandatory column is missing 'Spool Name'. 
The first column has headline Spool Name. 
[image]
I do not get it. As a minimum I could create a .csv (in Excel) with only one column Header Spool Name and Rows with different names. What do I make wrong?

OllisGit, 2020-08-24:
Hi @molodeztom!

Please attach your CSV-File to this issue so that I can take a closer look. 

What System to you use: Linux, Win, Mac, Browser, OP-Version, Python?

molodeztom, 2020-08-25:
Hi ,
it works now. SpoolManager1.csv works ..2.csv not. You may have a look but for me it is ok now. 
[Documents.zip](https://github.com/OllisGit/OctoPrint-SpoolManager/files/5122849/Documents.zip)
I have entered serveral spools with the UI and it works fine.
I use Windows10 Excel to edit the file and Raspbi 3B+ with octopi Image.
Thank You is a really helpful plugin.
Thomas

closed 2020-08-25 by molodeztom