OllisGit/OctoPrint-SpoolManager #39: Multi-printer support

KenLucke, 2020-08-25:
As with every other "filament manager" plugin, this leaves those of us with several printers between which we swap rolls of filament out in the cold.  This only manages spools on a single printer.  I have ~80+ different rolls of filament (just the currently open ones) that I use interchangeably between 4 different printers (with plans for two more).

Please don't suggest that I export and import .csv files every time I swap a roll in or out.  What a hassle.

Why cannot it be that a single database be used, on the computer that is running the browser(s) for the OctoPrint instance?  Specify a path to where the database would be stored on the local computer (not the Pi) and have all OctoPrint instances access that one database if desired.  Then, it could even be stored in a Dropbox or GoogleDrive [::gag::] folder which cold be accessed from any computer authorized to access that folder.  Yes, you'd have to have Busy/Locked/Unlocked flags on the file so that the database could not be changed while another instance was accessing it for a write, but that's just standard programming procedure.

Why can this not be done?  I've begged the authors of other filament management systems to do it, and no one ever does.

mrpegman, 2020-08-25:
Filament Manager does allow you to configure a central database to manage your spools.  It was a bit of a challenge to set up with PostgreSQL.  I've got 3 printers and it works pretty well once it's running.  The downside is that Filament Manager isn't being updated anymore.  I was excited to see this new plugin for managing spools pop up.  I'm waiting for the central database feature for this one.  It would be worth $ to me.

Jbloomer92, 2020-08-25:
I would love this feature, have 3 printers myself.

apresa, 2020-08-25:
+1 
support for multiple printers it's a must for serious makers

Doprintityourself, 2020-08-26:
+1 
As I have tonnes of spools and 10+ printers attached to octopi I would really appreciate to have it running on a database on one octopi that it is accessable to any other octopi instances in the network.
Thanks  in advance
appreciate your work!

OllisGit, 2020-08-26:
Hi @KenLucke!

>Why can this not be done?

I don't know, who said that this is not possible ;-) Of course, all your mentioned features could be possible!

Currently I am working on two major tasks: 1) integration with PrintJobHistory and 2) externalise the Database.
Number two is also needed in PrintJobHistory, but more important for SpoolManager (I know). It also include the feature to handle multiple OP-connections.
See Backlog: https://github.com/OllisGit/OctoPrint-SpoolManager/projects/1 

Kind of concept could be found here: https://github.com/OllisGit/OctoPrint-PrintJobHistory/wiki/Roadmap-to-an-external-Database
So, I want to implement a Database-Layer that I can use in both and future plugins.

> ...but that's just standard programming procedure

Yes, it is not "rocket science". Patterns/Concepts already exists.....but that doesn't mean that the implementation
 is done by a wink of an eye. 
For each feature (1. Dropbox, 2.GoogleDrive, 3.NAS-Folder, 4. Write-Lock, ....) you need to implement a lot of stuff:
- multiple UIs
- technical connection
- multiple authentification methods
- testcases
- ...

And you make some kind of "contract" after you release a feature. You need to maintain it (issue management, enhancements, discussions)!

Just asking ""why this, why that" is easy ;-)

Btw. the plugin development is done in my spare-time (one man show) and I have a couple of popular plugins I need to maintain.
So I have to prioritize hard!

Again, I am working on that feature, but be patient!

I will update the progress in this issue. Maybe I will also do some pre-releases/devel-branches, so that you can test very early the database feature....I only have one printer and not a farm. I can only do "virtual" testing. 
Testing by the community is highly appreciated.

Doprintityourself, 2020-08-26:
one reasons why I invested in the filament manager was the integration into octofarm
Is there I chance to make it work with octofarm?
Or would this be or to adresse to notExpected yet?

Doprintityourself, 2020-08-26:
By the way : Thank you for doing that at all and respect that you do it in spare time,

Appreciate your WORK!

OllisGit, 2020-08-26:
Hi @Doprintityourself,
thanks!

Regaring Octofarm: I never used OctoFarm and I have no idea how the integration with FilamentManager is done.
Maybe:
- Octofarm just connect to the external db with the "FilamentManager-Scheme"
or
- Octofarm is using a REST-API to collect the values

In both cases Octofarm needs be enhanced to communication with SpoolManager as well.
Maybe you can raise the question in the Octofarm issue tracker.

mrpegman, 2020-08-26:
I appreciate all the work that you're doing @OllisGit!  I'm a programmer myself and I can understand where your coming from.  The answer to "Can I?" is often yes, but the "How long?" is depends on a lot of variables.  Keep up the good work!

Doprintityourself, 2020-08-26:
Yes I will do  so,

Thanks for the quick answer and have a nice day !

Regards

Kai

OllisGit, 2020-08-26:
FUN: Welcome to my real life: 
"The Expert: Draw redlines with blue ink"
https://www.youtube.com/watch?v=47RI3V1z6pM

dorfman2, 2020-08-29:
@OllisGit - I'm excited for this feature to be released, thank you for your hard work! I am a Software QA tester for a living these days, so if you need an Alpha tester, please don't hesitate to reach out.

My use case is managing 4 OctoPi (more to come) instances for a Maker Space. This would allow us to track filament usage much better than before. We currently manually enter into a spreadsheet.

NotExpectedYet, 2020-09-16:
> Hi @Doprintityourself,
> thanks!
>
> Regaring Octofarm: I never used OctoFarm and I have no idea how the integration with FilamentManager is done.
> Maybe:
>
> * Octofarm just connect to the external db with [...]

Just to chime in for OctoFarm, I use the API that's revealed by FilamentManager and basically do a global call to the first online printer to sync the database with OctoFarms. Then any filament command I replicate on OctoFarm, and the specific OctoPrint instance with an API call. 

For this to work with OctoFarm, I would need external API access as I have no other access to OctoPrint currently. 

To support this I would need API endpoints, and possibly the database you have already planned. 

Personally the technical debt of supporting two filament manager plugins would be too much as I too am I one man band but I'm happy to try and work out some form of collaboration if it's beneficial.

OllisGit, 2020-09-18:
Just for your information: I made some progress implementing the external database connection:

https://github.com/OllisGit/OctoPrint-SpoolManager/tree/feature/externalDatabase

https://github.com/OllisGit/OctoPrint-SpoolManager/wiki/Roadmap-to-an-external-Database

The core setup is done, but there is a lot more to do. Don't use this version in production! 
This version only includes theses features:
- connect to postgres db (mysql driver is missing, need to find the right one)
- detects scheme version
- create new postgres database 
- stores/load spool values
- ..and a lot of issues und hickups ;-)

KenLucke, 2020-09-18:
Good to see progress...  often, developers are open to begin with, then things just die.  Thank you for your work.

moong8te, 2020-10-17:
Pretty happy to have found this discussion after already messaging Olli about this. 

When I felt the need to manage my spools I had a quick look at "Filament and Spool Manager". I personally like "Spool Manager" better. I can access it from the first screen and it gives me much more additional information.

After buying a second printer I do a lot of copy and paste lately.

Even more so after I wanted to have the Print Job History running which tells you to install all those additional plug-ins like "Filament Manager".

Not being sure if one of the filament managers might replace the other I started out populating both of them before every print. So now I have twice the manual work :-D

Database integration would be so nice.
In my case I have one for each printer as they are apart from each other.
This would mean I would need to turn them both on if the second printer wants to access the DB on the "masters' printers Pi.

Maybe that is the reason why Olli is looking into a cloud solution but as he pointed out there are so many services out there that one could imagine users would be quick to ask that their favorite cloud needs to be implemented as well.

In any case I am pretty happy about this thread. Looking forward to any kind of synchronization solution for "Spool Manager".

An once Spool Manager can be used in conjunction with "Print Job History" I will be super happy.

ldiegos, 2021-07-27:
> Personally the technical debt of supporting two filament manager plugins would be too much as I too am a one man band but I'm happy to try and work out some form of collaboration if it's beneficial to us both and the users.

@NotExpectedYet  In my opinión, if Spools Manager starts to communicate with octofarm, and starts using an external database(postgreSQL if you ask me) the old and discontinued Filament Manager could be uninstall from the octoprint instances.

So you don't need to maintain(improve) both API comms.

@OllisGit could mark the filament manager as discontinued given the chance to install the spools manager, he has the ability to do it since he maintain both.

Kind regards,

speedysk1, 2021-10-16:
Any updates on this?  I'm in the same boat and it would be nice.  I do realize you all work on this in your spare time.  Thanks for your efforts!  The plugin is awesome!

davidzwa, 2021-11-03:
Hey guys, I just wanted to add something to the discussion. I've been working on 3D Hub (https://github.com/3d-hub/3d-hub) which is a hard fork of OctoFarm. Currently working on a totally new filament management feature and I will take SpoolManager into account as a primary source of filament tracking, although it seems very beneficial to not add an external database as 3D Hub is already equipped with it.

Instead I'm leaning towards pushing the data through the connector plugin
https://github.com/3d-hub/3d-hub-connector

Let me know if this is something you'd like to see!

RobeeeJay, 2022-05-19:
I'm wondering if there has been any progress on support for an external database? I type this as I sit and watch my filament rolls go slowly out of sync across two Octoprint installations. 😭 (Thank you for your plugin!)

murray484, 2022-06-15:
Any updates on external databases?  I'm currently setting up Filament Manager because I need the shared database but a lot of the features I'd like to use are marked as "wontfix" on Filament Manager because they're already in Spool Manager, which I can't use.

ldiegos, 2022-10-15:
Hi there @OllisGit 
How is the external database going on?

BalooDK, 2022-10-15:
> Hi there @OllisGit
> How is the external database going on?

I don’t think anything is going on with this plug-in. It’s dead

mtinman, 2024-01-07:
Anyone here know of anything current (Jan. 2024) regarding this? Thanks in advance for reply! I think this is a GREAT idea, and would be VERY much appreciated by us makers with multiple printers/spool assets...

mtinman, 2025-07-05:
Hey All! @OllisGit, I just wanted to say that I am interested in helping with and/or reviving this part of the project, I just need to know how to go about it. Please let me know how!: mailto:j.m.wall@thegeekden.net, Octoprint Discord: WillyWonkyWun