OllisGit/OctoPrint-SpoolManager #40: Color editing

NameOfTheDragon, 2020-08-25:
I found a few issues with picking filament colors.

- Entering the hexadecimal color code doesn't seem to work and results in black, which means you can't copy-paste from another spool.
- When entering the _advanced_ color picker, it is not initialized with the current values from the spool, which makes it hard to work with.
- It would be nice to be able to define named colors and pick from the list. For example, Sunlu has "Pure Yellow" and "Red Copper".
- It might also be useful to be able to pick from the list of standard web colors, e.g. "Pale Goldenrod", "Dark Sea Green", etc.

OllisGit, 2020-08-26:
Hi @NameOfTheDragon,

1) Which browser do you use? I couldn't reproduce the behaviour with Chrome. 
When I click in the #-Textfield the Color Picker appears and everytime I entered a new hex value it is reflected to the preview box
2) I need to check if it is possible with the used Color-Picker lib (https://github.com/lauren/pick-a-color/). 
3) Good idea, maybe not a major point but a nice to have. I will put this to my backlog
4) If No3 is implemented you can define your own colors (maybe also the option to group custom colors as a pallete)
5) IMHO the used color-picker lib doesn't support such feature, it takes to much effort to implement vs. usage. Not on my roadmap. As a workaround you can use the Material oder Description field.

NameOfTheDragon, 2020-08-26:
Hi there. I've reproduced some of the issues in Chrome, although I think the first one might be partly due to a user-input error, but there is still something odd going on there. I thought the easiest way to handle this was to make a short video showing the issues, so here it is...
https://youtu.be/r19yntr29VM

OllisGit, 2020-08-26:
Hi @NameOfTheDragon,

Wow!!! Nice work, I should make a YT-Video as a mandatory information for raising an issue. Very helpfull!

Here are my remarks: 
- The SpoolManager integration is currently in progress. You can set the checkbox in the dialog and the popup will not appear anymore
- As already mentioned by you: You entered the blue color with a starting # sign, this is not needed (already there). The preview in the dialog is fine, but after saving in the DB there is a colorcode with two #-signs and the preview in the table is wrong. I need to fix this....but just enter the colorcode without #-sign
- I didn't find a way to pre select the advanced color, maybe I will turn on the spectrum for each basic color like this:
[image]

OllisGit, 2020-09-01:
Hi @NameOfTheDragon,

in the latest version 1.1.0 the current color is now reflected to the advance section. So. adjusting the brightness should be possible.

Please test and give feedback.

Thx, in advance
Olli

NameOfTheDragon, 2020-10-01:
Thank you. I did try this and it does work as expected. However I have temporarily disabled the plugin until there is proper integration with Print Job History.

Best regards,
Tim

closed 2020-10-02 by OllisGit