OllisGit/OctoPrint-SpoolManager #44: In the files list show filament usage in grams instead of volume

eraxar, 2020-08-27:
Hi there, in the old Filament Manager it shows the filament usage in the files list in grams instead of volume (see screenshot) may you can add this to the spools manager?
[image]

OllisGit, 2020-09-03:
fyi: will be added during implementation of "multi-tool support"

closed 2022-01-07 by OllisGit

TheHolyRoger, 2022-07-19:
Hey @OllisGit was this removed? I don't see the estimated grams in the files list, only in the "loaded" file. The old filament manager plugin used to show the grams for every file (when you click the arrow to expand)

B1ackCr0, 2022-11-11:
Just came here looking for the same thing.  I don't see it working on v1.7