OllisGit/OctoPrint-SpoolManager #46: Feature Request MMU2 aka MultiTool Support

kalleknall, 2020-08-28:
Multitool Support is needed to replace the old Plugin. It is so great to have 5 Filaments loaded for single Material prints.

Please plan it for short term devlopment. I think the most MMU Users needs this Feature.

Great Work so far !!!

codingcatgirl, 2020-08-29:
Yes! Please add this!

This plugin is awesome and i've been waiting something to finally replace the filament manager. However, i'll start using an MMU in a few days, so there is no use in migrating to this plugin for me yet if it won't support multi-material prints.

Looking forward to this :)

Zinip, 2020-09-05:
As soon as this gets this multi-tool/MMU2 suuport I'm going to be all over over, I can't wait! 

> Multitool Support is needed to replace the old Plugin. It is so great to have 5 Filaments loaded for single Material prints.
> 
> Please plan it for short term devlopment. I think the most MMU Users needs this Feature.
> 
> Great Work so far !!!

I can't agree with this any more... I'll be keeping on eye out for this feature to get put in, because once it does I'm going to be all over this. I haven't been this excited for a Plug-in since finding Filament Manger like well over a year ago!

comm, 2020-09-14:
Definitely support this request - I was enjoying using this plugin before I got my MMU unit but have had to stop. Happy to support - fundamentally its linking the "Extruders" defined in Octoprint to a spool.

Really look forward to this being implemented ASAP.

rickyh777, 2020-09-22:
I too would love to see this! For now I manually select when I have it in single spool mode but disable when doing multi prints. Otherwise great plugin!

codingcatgirl, 2020-10-22:
Stalebot being unproductive and problematic as usual. Do i have to set a reminder to post in this issue ever 29 days or can you please disable it?

codingcatgirl, 2020-11-15:
Good morning, before stalebot markes this issue for auto close again, i'd like to say that i'm still very much hoping that this will be implemented eventually. :)

Zinip, 2020-12-01:
I'll tackle this bi-weekly bump, as I also have continued hope for this! ...One might even be able coerce some begging out of me, if absolutely necessary. 😅

closed 2021-01-10 by stale[bot]

OllisGit, 2021-01-24:
fyi: this is the next big thing (in front of "external database"")

codingcatgirl, 2021-05-14:
Since i really needed this feature right now, i decided to just implement it myself :) I have most parts already done and only need to adjust a few things before i can submit a pull request – should be done by tomorrow or at least this weekend!

codingcatgirl, 2021-05-15:
Here we go: Please test :) #161

PR #161 (2021-05-19): Multi-Tool/Multi-Material support
Hi! 

Since i really needed Multi-Tool support in this Plugin right now, i decided to just implement it myself :)

Simply put: It's done and this is how it looks: 

[image]

List of Changes/Features:
- Revamped entire backend to work with spools for multiple tools (fixes #46, #153)
- Allows manual filament-change mid-print. If a new spool gets selected mid-print, the user gets asked whether they would like to commit the filament usage of the print so far to the previously selected spool.  (fixes #130)
- Ensures that a spool can't be selected in multiple tools
 [... cut]