OllisGit/OctoPrint-SpoolManager #47: Change Name in Settings from "SpoolManager Plugin" to "Spool Manager"

codingcatgirl, 2020-08-29:
For some reason, in the Octoprint settings, the word "Plugin" is prepended is to the name of the Settings section associated with it.

That seems a bit pointless and redundant to me and i assume it's a mistake :D Please change it.

OllisGit, 2020-08-30:
Hi @codingcatgirl ,

did you test your PR? Your change has no effect to the settings menu.

I looked a little bit deeper  into the OP-Code and found the issue. The menu use the plugin_name variable.... that was the reason why I didn't change it in the past for my other plugins.

BUT, I raised a PR for OctoPrint to provide customisation of the Linkname: 
https://github.com/OctoPrint/OctoPrint/pull/3710

Maybe we have success and the PR is excepted....will see!

codingcatgirl, 2020-08-31:
Oh wow, no i have to admit i didn't test it because it really didn't expect there to be any other possible cause. Thanks for taking care of it though :)

OllisGit, 2020-09-01:
Hi @codingcatgirl,

the name is now adjusted in version 1.1.0..

BR
Olli

closed 2020-09-01 by OllisGit

codingcatgirl, 2020-09-02:
I saw, awesome!