OllisGit/OctoPrint-SpoolManager #50: Spool utilization history?

Lumute, 2020-08-30:
Thanks for this great plugin!

It would be really cool if there was a way to see a log of every time material is subtracted from a spool, like date/time and amount used.

I understand you are working on integrating with PrintJobHistory so maybe that is already part of this planned integration where records on the spool log could also be associated to print job in the PrintJobHistory?

In any case, having a button in the spool to show the utilization history would be great, and a button to add manual adjustments that would registered in the utilization log.

OllisGit, 2020-08-31:
Hi @Lumute,
thats the final goal with the PJH integration.

But what do you mean with "manual adjustments that would registered in the utilization log"?
Can you give me an example?

Lumute, 2020-08-31:
Hi,

Awesome, sure for example I had a failed print that didn't register in PrintJobHistory due to a power failure, so if we have a proper spool log then I would click a button to do a manual adjustment of -7g (or 1.61m) with datetime and reason/description = "Failed print due to power outage", or "Gave 5m to a friend for testing", etc.

Ideally if I take the utilization log and all add the entries + the remaining spool I should get what the spool had when I started tracking...

Thank you!

marneu, 2020-09-22:
[current plan]https://github.com/OllisGit/OctoPrint-SpoolManager/wiki/rdbs-scheme-development
@Lumute: Could be solved by adding a record to SpoolHistory, is an additional text really needed?

Lumute, 2020-09-22:
You mean the explanation / note? not really needed but would be a nice to have...

marneu, 2020-09-22:
There could be a field "originator" which could be (ab)used with a comment, but limited to 16 chars.

Lumute, 2020-09-23:
I think that is a good compromise, maybe not a long note but at lest something short to remeber what was the adjustment for.

excalbian, 2020-10-12:
With PJH now supporting [adding jobs](https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/77) would that be a better option? You could make the adjustment by adding a job to PJH and note how much filament was used (consumed). Since PHJ already has a comments section, then you could describe in lots of detail what the filament was 'used' for.

OllisGit, 2021-01-17:
fyi: I am continuing working on this in Q1/2021

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

OllisGit, 2022-01-16:
fyi: Not closed any more, because I want to implement this in one of the next releases.