OllisGit/OctoPrint-SpoolManager #52: Enhancement: List of Material

sflebrun, 2020-09-02:
Installed the Spool Manager plugin today.  The list of materials does not include PLA+.  I use both PLA and PLA+ filaments and it would be nice to track the difference in the database.

Is it possible for an End User to add to the list of possible materials?

If not, could PLA+ be added to the list?

-- 
  Steven F. LeBrun

Quote: "Behold the lowly turtle," the astronaut had quoted. "He only makes progress when he sticks his neck out."
     -- Ben Bova, from "Return to Mars"

OllisGit, 2020-09-02:
Hi @sflebrun ,

just enter the new Materialname into the field and you are done. The name is stored in the database for that spool and also the name is suggested during edit of other spools.
But there is no relationship between the new material name and the density. You need to enter the density by your own.

BR
Olli

sflebrun, 2020-09-03:
Thanks.  I missed the ability to provide your own material name.  I missed the dialog box when I first added a spool to the database.

OllisGit, 2020-09-03:
Added in latest version 1.2.0

Please test and give feedback

Thx, in advance
Olli

Lerxst151, 2020-09-03:
I just installed 1.2.0 and I do not see PLA+ in the list

sflebrun, 2020-09-03:
I manually updated to version 1.2.0.  I did not see any difference.  No default density was filled in for PLA+.  The problem, I suspect, is that I added PLA+ in version 1.1 and provided no default density.  The upgrade to 1.2.0 did not overwrite my previous settings.  Is the list of material editable by the end-user?  If so, I can update it with the default data for PLA+.

jmarc67, 2020-09-04:
Hello

Same as sflebrun. No PLA+ material added with 1.2.0

OllisGit, 2020-09-04:
Upps..didn't merge this feature, sorry! PLA plus is not in release 1.2.0.

Currently the List with Material/Density is hardcoded here: https://github.com/OllisGit/OctoPrint-SpoolManager/blob/master/octoprint_SpoolManager/static/js/SpoolManager-EditSpoolDialog.js#L63

It is not planed to have a userdriven material-editor...maybe a long term activity.

SM-26, 2021-04-14:
I'm not sure if it's something I did, But I've got PLA+ twice
[image]

closed 2022-02-28 by github-actions[bot]