OllisGit/OctoPrint-SpoolManager #53: Feature Request: Tolerance of Filament

sflebrun, 2020-09-02:
I would like to request that the tolerance of filament be tracked in the database along with diameter.  This information could be useful in tracking performance of a filament.

For now, that information could be stored in the comments though it makes it difficult to query the database for that information later.

-- 
  Steven F. LeBrun

Quote: "The objection to fairy stories is that they tell children there are dragons. But children have always known there are dragons. Fairy stories tell children that dragons can be killed."
     -- G.K. Chesterton

OllisGit, 2020-09-02:
Need more details: 
- do you want to add % or float value?

sflebrun, 2020-09-03:
Manufacturers provide the tolerance as a floating point number such as +/- 0.02 mm and +/- 0.05 mm.  Therefore, IMO, storing the value as an absolute floating point number would be best.

OllisGit, 2020-09-03:
Added in latest version 1.2.0

Please test and give feedback.

Thx, in advance
Olli

sflebrun, 2020-09-03:
Manually updated to version 1.2.0.  The tolerance field works fine!  Thanks.

closed 2020-09-04 by OllisGit