OllisGit/OctoPrint-SpoolManager #54: No 1.1.0 in Software update

jmarc67, 2020-09-02:
Hello,

I cannot see 1.1.0 SpoolManager plugin in Software update.
No update even with "Force check for update" option

[image]

OllisGit, 2020-09-02:
I takes some time. Please do the following:

- Filter: Spool
- You will see version 1.0.0
- Open menu and click on "refresh"
- Now you can (should)see version 1.1.0 
[image]

jmarc67, 2020-09-02:
I've done that and:

[image]

Something strange because i had "Reinstall" and not "install"

closed 2020-09-02 by jmarc67