OllisGit/OctoPrint-SpoolManager #58: How to test the plugin without printing?

Spacemarine2018, 2020-09-06:
I'd like to test the plugin and see if it covers all aspects of my workflow and see if there are any bugs. But I don't want to do a ton actual printing.

Is there any way to simulate a print and simulate the usage of filament and watch the database update accordingly?

Also for testing purposes: Can I place a g-code command in my g-code that will alter the amount of filament stored, without using the G1 E command?

And I also didn't find any documentation on how this is implemented: How does the plugin know how much filament is used? Does it count only successful g-code commands? Or does it count all the G1 E commands in the g-code file? What happens if I try to do a long cold extrusion that get's prevented by octoprint or my printers firmware?

tduehr, 2020-09-07:
Way at the bottom of the OctoPrint settings is a virtual printer setting that will create a `VIRTUAL` serial port with an attached printer implementation.

OllisGit, 2020-09-07:
As @tduehr mentioned, use a virtual printer. 
Maybe this plugin helps you to add the v-printer to your environment:
https://plugins.octoprint.org/plugins/virtual_printerconfig/
> Note that this plugin is only compatible with OctoPrint 1.4.1 and higher.

Documentation is a good point. I added this to my "ShortTerm-Task list"

closed 2020-10-17 by stale[bot]