OllisGit/OctoPrint-SpoolManager #59: Temperature management

yceoshda, 2020-09-07:
Hi,

This is a feature request.

Is it possible to have the temperature actually sent to the printer (like filament manager does with it's offsets) either by default, or with an option. It'll really handy to be able to slice something once and then print it with a bunch of different spools without worrying of the temperature.

Thanks

PS: Thank you for your amazing work on those plugins of yours.

WyattNielsen, 2020-09-09:
This is the only feature that I needed to switch from filamentManager. I took a quick initial pass at the feature and opened a draft PR to advance the conversation.

WyattNielsen, 2020-10-16:
@OllisGit any thoughts on this feature? I'm willing to update the PR as needed.

posicat, 2020-11-12:
I'm currently using Filament Manager, mostly for the temperature feature.  Maybe it would help if I shared my use case.

I print mostly with PLA, and PETG, but I have a number of different filaments from different manufacturers, sometimes the bargain bin stuff I find on clearance, and some of it really nice high quality stuff.  I like to customize my per-type settings in Cura,   Then in Octoprint I share the database of filaments between 2 different printers (the Good one and the "experimental" one).  For each filament I tweak the sweet temperature spot by adjusting the delta in filament manager up or down a few degrees.

What would be ideal for my workflow is 

1) A delta number that's added or subtracted from the temperature set in the G-code.
2) A quick button on the GUI to get to those settings (for the filament I have selected) because I spend some time dialing it in on the printer, and then record my change in the filament manger.
3) It'd be nice if the delta was displayed on the GUI somewhere as well.

I know other people use the tool for managing how much of a spool is left, but it looks like you've already got that handled really well!  I tend to do small experimental stuff I design myself and am less concerned if I run out of filament because I'm only doing 15min-2hr prints.

I'm a software developer too, so feel free to poke at me if you need any more information, and I'm always happy to help test betas.

newphreak, 2020-11-29:
Is it possible to get the value from selected spool to some variable so i can use that in the octoprint start script? Even that little would be enough for me. Should be quite simple?

OllisGit, 2021-01-17:
Hi @yceoshda, @posicat,

I will take a look into the provided PR from @WyattNielsen. Take a look into the comments
https://github.com/OllisGit/OctoPrint-SpoolManager/pull/61

@newphreak I added a new issue for your request, if you have additional infos, just add it there:
https://github.com/OllisGit/OctoPrint-SpoolManager/issues/98

yceoshda, 2021-01-18:
Hi @OllisGit,

Thank you for looking into it! Sorry I can't really be of help, I don't know Python (and unfortunately have no time to learn it properly :-( ).

OllisGit, 2021-01-24:
Hi, 
im working on the PR, but because I never used the temp-offset feature I have no clue if it is working or not, see my comment in the PR https://github.com/OllisGit/OctoPrint-SpoolManager/pull/61#issuecomment-766093327

Any ideas?

OllisGit, 2021-05-24:
Hi,
in the latest release V1.4.0 the PR from @WyattNielsen  is now included. Thanks again!!!

Please test and give me a feedback.
Olli

yceoshda, 2021-05-28:
Hi, haven't had the time to fully test it, but it seems to work! Thank you very (very very) much to all of you for the work on this.

closed 2021-06-05 by OllisGit