OllisGit/OctoPrint-SpoolManager #62: Feature Request: start after x seconds

AJErazzor, 2020-09-15:
Would it be possible to throw up the warning about the filament color check for x seconds and if not changed then start the print?  I like having the check there - but often I am doing multiple prints on the same spool and keep forgetting to confirm only to come back and the print not started.

OllisGit, 2020-09-22:
You mean something like in my "Autostart-Plugin"
https://github.com/OllisGit/Octoprint-AutostartPrint
[image]

AJErazzor, 2020-09-22:
Yes exactly like that but driven from when I click print button - shows the message about the spool selected and then starts the countdown

closed 2020-11-01 by stale[bot]