OllisGit/OctoPrint-SpoolManager #63: Wrong sorting

spanzetta, 2020-09-20:
Pls take a look at the screenshot.. 
It's self explaining 

[image]

spanzetta, 2020-09-22:
Hi @OllisGit 
any comments?
Thanks

spanzetta, 2020-10-18:
It looks like no one is reading  it.. 
:-(

spanzetta, 2020-11-23:
Bug still there 
Can someone look at it?

OllisGit, 2021-01-17:
Hi @spanzetta ,
fyi: I will take a look on this in the days

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

OllisGit, 2021-01-24:
Hi @spanzetta,

sorry to say, but I can't reproduce the issue.
I use the database-command for sorting, maybe  some values are not detected as numbers and interpreted as text.

Could you take a look into your database? Do the following:
- copy your spooldatabase.db to your desktop pc (see storage tab for the exact location)
- open the database with a tool like: https://www.dbvis.com/ or https://sqlitebrowser.org/
- select the table 'spo_spoolmodel' and press on the 'remainingWeight-column' to do the sorting in the tool

or just attach the database to this issue and I can do the check for you

[image]

spanzetta, 2021-01-25:
Hi @OllisGit 
first of all thanks for putting your hands on this..

I was trying to do what you are asking but:
1) If it try to download the file from Octoprint interface (the blue button) I get the following error:
(it points to the page http://192.168.1.21/plugin/SpoolManager/downloadDatabase)
_Internal Server Error
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application._
2) If I try to download manually the file using FileZilla.. when trying to access to it I get the following error
_Stato:	Connessione a 192.168.1.21...
Risposta:	fzSftp started, protocol_version=9
Comando:	open "pi@192.168.1.21" 22
Stato:	Using username "pi". 
Comando:	Pass: *********
Stato:	Connected to 192.168.1.21
Errore:	FATAL ERROR: Received unexpected end-of-file from SFTP server
Errore:	Impossibile collegarsi al server_
so I cannot even download the DB manually to do the requested check..

But pls note that Octoprint works fine trough web interface and that SSH is enabled as I can access to it using Putty..
see below login output

_login as: pi
pi@192.168.1.21's password:
Last login: Mon Jan 25 16:53:47 2021 from 192.168.1.23

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

------------------------------------------------------------------------------
Access OctoPrint from a web browser on your network by navigating to any of:

    http://octopi.local
    http://192.168.1.21

https is also available, with a self-signed certificate.
------------------------------------------------------------------------------
OctoPrint version : 1.5.3
OctoPi version    : 0.17.0
------------------------------------------------------------------------------

pi@octopi:~ $_

What's wrong? Can you pls help me?

Thanks

PS I have everything updated.. see all versions below
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:84.0) Gecko/20100101 Firefox/84.0
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1400
env.hardware.ram: 385024000
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 3 Model A Plus Rev 1.0
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 19.3.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3
printer.firmware: Marlin Creality 3D

OllisGit, 2021-01-25:
Hi @spanzetta ,

the download issue should be solved, but it is not in the offical release.

But you can install the dev-version from here: https://github.com/OllisGit/OctoPrint-SpoolManager/archive/master.zip

Btw. did you install your sftp server on the raspi? I think this is not a default service that runs out of the box.

spanzetta, 2021-01-25:
ok.. thanks.. I will try using the dev-version..
About stfp server.. no I dind't install anything.. can you pls quickly guide me how to do?
Thanks

spanzetta, 2021-01-25:
I downloaded the database and analyzed with SQL Lite browser..  and..
Inside the SQL browser data looks fine.. and I can sort them..
But inside Spool Manager the sort function return wrong sorting..
See both screenshots
Pls check also the attached database too (to me it seems that data are different...)
[spoolmanager_database.zip](https://github.com/OllisGit/OctoPrint-SpoolManager/files/5868762/spoolmanager_database.zip)
[image]
[image]

spanzetta, 2021-01-27:
Hi @OllisGit did you have the chance to look at the database to understand why there is this problem?
Thanks in advance

OllisGit, 2021-02-02:
Hi @spanzetta,

I think I found the issue. You imported the weight via csv-import, right?
 
Unfortunately the plugin is also calculating the weight in the UI....current workaround could be: open each spool and press save.

spanzetta, 2021-02-02:
Hmm .. I don't remember..
I will anyway follow your suggestion (open each spool and save) and see if something changes 
Thanks a lot

spanzetta, 2021-02-02:
Solved! The trick fixed the problem..
Case closed!
Thanks a lot

closed 2021-02-02 by spanzetta