OllisGit/OctoPrint-SpoolManager #64: Feature Request: Change or add "Filament weight" calculation & percentage

killwack, 2020-09-21:
Hi OllisGit.

Thank you in designing such a great tool and your efforts to do so. I use your latest Spool Manager tool 1.2 now with almost 20 different filaments to manage my spools.

The only issue I have at the moment is the existing calculation of the displayed weight of each spool.

I am missing the value /percentage of the filament weight itself, which at the moment seems not to be calculated in my unterstanding of the shown valuses.

See my example below and the yellow marked weight calculation currently used.

* I have a 1000g (1kg) filament PLA spool in black. 
* The spool from this manufacturer itself has a weight of exact 207g 
* My first print job using this spool used in total 195.3g.

What I am missing now is a display of the remaining weight of the filament itself, not the remaining weight based on the total weight including the spool weight which seems to be currently calculated.

The so called "filament remaining weight" would be more important for me. 
According to my example this should be 

"filament remaining weight" = 1227.0 (total weight) - 207.0 (spool weight) - 195,3g (used weight) = 824,7g including the correct percentage depending on this calculation formula.

[image]

It would be very nice if you could add this form of calculation in your overview as a value which can be displayed in the overview section and detailed section as shown in my screenshot.

Thank you and best wishes, Rainer

5teveO, 2020-09-22:
I agree with this also. It seems a bit unintuitive having it this way. I mentioned it in another thread. 

I also thing Auto length calculation would be possible as long as density and diameter are there. Having the option to override the calculated length if known exactly.

I also very much agree its a great plugin!

OllisGit, 2021-01-17:
Hi @killwack , @5teveO ,
totally agree! The Spool weight should be subtracted ....but currently I have some issues/challenges about autoupdate calculates values.
This issue is almost the same as this #79
So, in the spool-weight case: What is the best event to recalculate the remaining value? After adjusting the spool weight or also if I change total weight.
It is not so easy..there could be some circle reference-issues.

Maybe you can make a "definition" when and how a value should be autochanged or make a suggesting of an UI (Buttons or readonly)

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

OllisGit, 2022-01-07:
related to PR #206

OllisGit, 2022-01-12:
Hi! 
The mentioned PR is now in the latest release!

Because I have a lot of issues related the weight calculation I close this issue to get a more "up to date overview" in the issue-tracker.
If you found an issue in the latest version feel free to raise a new issue!

closed 2022-01-12 by OllisGit