OllisGit/OctoPrint-SpoolManager #65: Where do I select a Spool for printing

war1000, 2020-09-22:
Hi, this is a bit of headscratcher for me. I put in the spool info to the best of my knowledge but where do I select the spool? When I try to print, it does give me the warning that a spool has not been selected and if I should proceed without one. But I can't find the option to select the spool. Can you please point it out to me? Thanks!

[image]

5teveO, 2020-09-22:
Bottom Left hand corner of the Octoprint Main page..  under spool manager plugin, under printer notifications!

ThomDietrich, 2020-10-14:
Recommend to install https://plugins.octoprint.org/plugins/dragon_order then Drag&Drop the Dialog further up

closed 2020-11-23 by stale[bot]