OllisGit/OctoPrint-SpoolManager #67: verify selected spool

slippy115, 2020-09-23:
keeps telling me to verify selected spool  how do i do that

Yang-Richard, 2020-09-30:
I'm also curious how to do this. I can't find a settings page or area to verify the selected spool, even though I have a spool selected and it automatically decreases the remaining spool percentage after prints.

ThomDietrich, 2020-10-14:
This is just an unfortunately phrased request to confirm your selection. See #74

ThomDietrich, 2020-11-24:
@OllisGit is stalebot closing tickets in your absence really intended? The bot became quite popular over the years but I am convinced that in small less active repositories it does more harm than good.

OllisGit, 2021-01-17:
Hi @slippy115, @Yang-Richard,

the PR from @ThomDietrich will be in the next release.
  
PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

OllisGit, 2021-01-24:
Hi,

PR is implemented in latest release 1.3.0.

Please test and give feedback.

Thx, in advance
Olli

closed 2021-05-19 by OllisGit