OllisGit/OctoPrint-SpoolManager #70: Some spool colors do not showup in spool list

theone11, 2020-10-04:
When choosing custom color #xxxxxx in spool information page **most of the times** the color doesn't show up i the spool list.
It **always** shows up in the spool inforamtion page

For example when chossing color LightSkyBlue #87CEFA it does show up in spool information:
[image]
but not in spool list:
[image]

and for another custom color DarkVoilet $9400D3:
[image]
It does show up:
[image]

All my color definitions where taken from
https://www.w3schools.com/colors/colors_groups.asp

OllisGit, 2020-10-09:
Hi @theone11 ,
you copy and pasted the color-value plus the starting # in the first picture. So, in the database we have two hash-signs:

[image]

I need to fix this behaviour.

OllisGit, 2021-01-24:
Hi @theone11,
fix is in lates release v1.3.0

BR
Olli

closed 2021-01-24 by OllisGit