OllisGit/OctoPrint-SpoolManager #71: "save" button is not accessible

skimaniac, 2020-10-08:
The "save" button is not accessible in normal screen mode on a macbook pro 13 ". The box to add comments is displayed over the button and prevents pressing" save "we must absolutely switch to full screen mode .

Normal screen
[image]

Full screen
[image]

OllisGit, 2020-10-09:
Hi @skimaniac ,

I could reproduce the issue on my mac with safari, but it didnt' appear with chrome. The layout is broken, but the save button works.
In need to analyse this. In the meantime, try to switch to an other browser or change the resolution.

OllisGit, 2021-01-24:
fyi: I did a small improvement maybe this will fix the issue on your side. It will be in the next release.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

OllisGit, 2021-01-24:
Hi, @skimaniac,

release 1.3.0 is out, please test and give feedback.

Thx, in advance
Olli

closed 2021-05-19 by OllisGit