OllisGit/OctoPrint-SpoolManager #72: WARNING: Filament prediction not possible! One of the needed fields are not a number in Spool

AndersonGER, 2020-10-10:
Hello,

i use the Plugin since a few Days, and insert all my Spools already.
When i try to print, i got the Issue above displayed.
I´ve already had a look @_♦ https://bitfab.io/blog/3d-printing-materials-densities/_ to improve that i got all Fields filled correctly, but it seems something is still wrong.

Can you  please help me to identify my fault?
My Fields are actual filled like this:
 
[image]

Best Regards and lot of thx for your help
Anderson

skimaniac, 2020-10-10:
you must not put a comma (1,75. 0,1) for the decimal, you must use the dot (1.75  0.1)
[image]
.

AndersonGER, 2020-10-11:
Hello Skimanic,

this fixed my Problem, thank you very much :)

Best regards from Germany
Anderson

closed 2020-10-11 by AndersonGER