OllisGit/OctoPrint-SpoolManager #73: Spool manager generate a lot of errors

courtpl, 2020-10-11:
Spool manager generate a lot of errors, and cause a freeze of the octoprint server.
See error log
[octoprint.zip](https://github.com/OllisGit/OctoPrint-SpoolManager/files/5361376/octoprint.zip)

OllisGit, 2021-01-17:
Hi @courtpl,

it looks like that there is something wrong with the tool-setup. Do you use a multi-extruder printer?
Could you please provide a sample gcode where the issue appeared (e.g. Galaxy_Buds_Holster_2_v3_v0.3_4mf_0.1mm_PLA_MK3SMMU2S_6h22m.gcode)

I think there is also a python bug.

All bugs
```
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager/__init__.py", line 349, in _on_file_selectionChanged
    allFilemants = metadata["analysis"]["filament"]
KeyError: 'filament'

  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager/Odometer.py", line 72, in parse
    for i in xrange(len(self.lastExtrusion), self.currentTool + 1):
NameError: name 'xrange' is not defined

  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager/Odometer.py", line 62, in parse
    self.lastExtrusion[self.currentTool] = e
IndexError: list assignment index out of range
```

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

OllisGit, 2021-01-24:
Hi @courtpl,,
fixed in latest release v1.3.0.

BR
Olli

closed 2021-01-24 by OllisGit

courtpl, 2021-01-25:
Thanks for the job.
Regards