OllisGit/OctoPrint-SpoolManager #75: A bit confused Should i use this or filament manager?

pixelicous, 2020-10-17:
Hey,

New to octoprint etc..
Is there any difference between spoolmanager and filament manager?
Am I missing something and they complete different thing? It seems to me like they do the same thing

So which one should i use? It seems like both are being updated..

moong8te, 2020-10-18:
Hello pixelicous,

I'm also rather new to the situation but as far as I understand the plug-ins do have quite some similarities.
The difference seems to be that originally they had been created by two different developers.

As far as I found out Spool Manager was created by Olli and Filament Manager was created by somebody else - possibly by malnvenshorn?

It seems that the Filament Manager project became abandoned at some point and was then adopted by Olli who already had experience in this field and is otherwise very active for octoprint which is much appreciated.

In addition Filament Manager seems to be part of a whole plug-in suite. I also use "Print Job History" which is then asking you to install Filament Manager and other plug-ins.

Reading through this git it seems that Olli is working on the integration of Spool Manager to this Print Job History plug-in. This would mean that in the future users would have the choice which filament management plug-in can be used with Print Job History.

From my point of view I am really looking forward to this integration as I like Spool Manager much better. I gives you more information (first use of spool, last time used etc) and it is accessible directly from the first page without having to dive into the configuration to manage your filament.

But as I said I am also still in the process to understand the details around this and maybe not everything above is 100% correct. So take it with a grain of salt.

Best Moong8te

pixelicous, 2020-10-20:
@moong8te thanks for the elaborate response man.. i was sure ollie would respond first  :)
I haven't heard of print job history.. will take a look at it later.

so i think ill keep on using spool manager

thanks for clearing things up

closed 2020-10-20 by pixelicous

OllisGit, 2021-01-17:
Hi @moong8te ,

great explanation! Totally right!
After some bug fixing in my other Octoprint-Plugins and finishing the external database feature this is the next in my list (see roadmap https://github.com/OllisGit/OctoPrint-PrintJobHistory/projects/1)

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.