OllisGit/OctoPrint-SpoolManager #78: [FEATURE REQUEST] Add manual print entry (with used weight/length)

felislynx-silae, 2020-10-20:
Hi,

I'm using octoprint mainly as camera + percentage viewer. And i'm starting prints from printer directly.
Could you add button that allows to manually add entry of spool usage (for example taken from slicer). It would be very neat for someone like me.

Cheers

OllisGit, 2020-11-03:
Hi @felislynx-silae, 
hmm.. I am not sure what you mean? Because you can update the usage any time.

Please explain with more details.

felislynx-silae, 2020-11-03:
Yes manually clicking on spool and substracting usage from cura. I would like to have button where i choose spool, insert how much grams/meters used and it updates last time printed and remaining weight

OllisGit, 2021-01-17:
Hi @felislynx-silae,

how can I receive the value from cura? Is there an API to get that information?
I am still a little bit lost about your request. Maybe an example would help.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

OllisGit, 2021-04-15:
No feedback, if needed raise new ticket.

closed 2021-04-15 by OllisGit