OllisGit/OctoPrint-SpoolManager #79: Feature request: When updating total weight, automatically update length

pixelicous, 2020-10-20:
If you forget to update your spool, then you go ahead and update it manually, the weight is updated but the length used isn't updated automatically.. which is a bit weird because for any other change the system does, it does it.

Would be great to have it done automatically..

ThomDietrich, 2020-11-19:
I agree. Generally it would be nice if data entered for a spool is used to fill the remaining fields

OllisGit, 2021-01-17:
Hi @pixelicous,
thanks for the request. Automatically updating fields is a tricky thing.

We have a couple of fields that have a "relationship", but what is the best strategy for autoupdate?

If I enter/adjust the used weight the used length should calculated, right? But what about the other way around.
Adjust the used length should the used weight also update (then we hava a circle)...which field is the "master"?

Maybe a button on each filed that can be execute the calculation of the current values, but then I think the UI is  a little bit "overloaded".

Do you have an idea how to solve this or a suggestion for the UI?

BR
Olli
PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

Sembazuru, 2021-01-31:
> If I enter/adjust the used weight the used length should calculated, right? But what about the other way around.
> Adjust the used length should the used weight also update (then we hava a circle)...which field is the "master"?

Technically, isn't the used length really the master during usage? AFAIK scrubbing the gcode for usage really only shows length (the g-code for extrude is in unit of length, not weight or even volume). So when automatically updating the usage the original data is length. Used weight is calculated from length based on entered density and diameter (used to calculate cross-sectional area) in OctoPrint either by g-code parsing before printing or g-code capture at the serial port. If there is are numbers provided by the slicer that the g-code parser reads, those are probably converted from volume of the tool path.

But, when entering a brand new spool, except for a few manufacturers the master field there is the weight. (Only a few manufacturers actually tell you the length of filament put on the spool.)

> Do you have an idea how to solve this or a suggestion for the UI?

For the pairs weight/length, my thought is one should be updated when the other is changed (or looses focus). The question in my mind is which one gets changed if either the density or the diameter get updated? Maybe both marked red until one gets and loses focus to update the other?

OllisGit, 2022-01-07:
releated to PR #206

OllisGit, 2022-01-12:
Hi! 
The mentioned PR is now in the latest release!

Because I have a lot of issues related the weight calculation I close this issue to get a more "up to date overview" in the issue-tracker.
If you found an issue in the latest version feel free to raise a new issue!

closed 2022-01-12 by OllisGit