OllisGit/OctoPrint-SpoolManager #80: Can't start print with plugin installed

jbrown17, 2020-10-23:
With this plugin installed, I'm not able to start a print or set a file as selected. I'm not getting any errors or visual feedback of any kind when clicking a file or the print button. It sits there as if nothing has happened.

I've narrowed it down to this plugin (uninstalling fixes the issue), but I didn't see any known issues or conflicting plugins.

I'm running Octopi 1.3.12 on a Raspberry Pi 3 B+. Let me know what other useful information I can provide!

OllisGit, 2020-11-03:
Hi @jbrown17 ,

which browser do you use? 
Can you open the browser developer console (most browsers use F12-Key to activate the developer tools).
Maybe you can find an error message in the dev-console.

jbrown17, 2020-11-03:
I'm using Chrome. And yep, there was an error. Here it is:

```Uncaught TypeError: Cannot read property 'permissions' of undefined
    at FilesViewModel.confirmSpoolSelectionOnLoadAndPrint [as loadFile] (packed_plugins.js?ed65e899:5873)
    at Object.eval (eval at parseBindingsString (packed_libs.js?ef1e2987:11374), <anonymous>:3:114)
    at HTMLDivElement.<anonymous> (packed_libs.js?ef1e2987:11395)
    at HTMLDivElement.dispatch (packed_libs.js?ef1e2987:5184)
    at HTMLDivElement.elemData.handle (packed_libs.js?ef1e2987:4992)
confirmSpoolSelectionOnLoadAndPrint @ packed_plugins.js?ed65e899:5873
eval @ VM4704:3
(anonymous) @ packed_libs.js?ef1e2987:11395
dispatch @ packed_libs.js?ef1e2987:5184
elemData.handle @ packed_libs.js?ef1e2987:4992```

OllisGit, 2021-01-17:
You mean Octoprint 1.3.12 and not Octopi, right?

 I looked into the sourcecode and unfortunately the compatibility with 1.3.x is broken since V1.1.0, sorry!

Could you upgrade to OP1.4.x or even better OP1.5.x

I will adjust the plugin description in the next reales that this plugin is not working with 1.3.x anymore.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

closed 2021-04-15 by OllisGit