OllisGit/OctoPrint-SpoolManager #81: Hide Empty spools shows spools that is more than 100% used

Pizzahd88, 2020-11-01:
If ex your spool has a bit more filament than normally there will be consumed ex. 1001g of 1000g (100,1%)

This spool will be shown when selecting "Hide Empty". "Hide Empty" should hide all spools with >= 100% consumption.

## Feature Request in this regard:
Could it be possible to manually mark a spool as empty, instead of correcting the usage amount?
Thereby if you lost a lot of filament due to ex filament changes etc, you could mark a 99% used spool as empty and keep the actual meassured usage instead of having to correct the remaining weight.

OllisGit, 2020-11-03:
Hi @Pizzahd88 ,

hmmm..adding a new attribute like "active" is maybe a better solution with more options.

What do you think?

Pizzahd88, 2020-11-03:
Hi @OllisGit 

Yea, marking spools as active and the option to show only active spools could be a better feature as we often won’t want to see the new spools still in the boxes as well as the spools that are empty

OllisGit, 2021-01-17:
PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

Pizzahd88, 2021-01-17:
I am just happy you make a product I get to use for free. 

Enjoy yourself and stay healthy my friend.

OllisGit, 2021-01-24:
Hi @Pizzahd88

new feature is implemented in release 1.3.0.

Please test and give feedback.

Thx, in advance
Olli

OllisGit, 2021-04-15:
No feedback -> closed!

closed 2021-04-15 by OllisGit