OllisGit/OctoPrint-SpoolManager #82: Suggestion for if/when spool scanning added

xrxca, 2020-11-03:
I'm probably in the minority but I've got a hand scanner linked to the machine I normally connect to octoprint from, if scanning gets implemented it would be nice to be able to use that hand scanner rather than capturing an image from the camera attached to the octoprint server.
I currently scan the UPC code and/or the Amazon barcode as part of the info I put into the info field, but if scanning does get implemented a clear field for the scanned code would be nice.

OllisGit, 2021-01-17:
Hi @xrxca ,

this feature is already on my list (see project. mid-term: https://github.com/OllisGit/OctoPrint-SpoolManager/projects/1)...but instead of supporting a lot of scanning hardware I was thinking about a Smartphone-App that scannes QR/Bar-Code and send the meta-data result to the local OP-Installation.
Coud that be an option for you? Or do you know something about a "standard-API" for HW-Scanner?

Btw. a QR-Generator-Feature is provides as a PR (https://github.com/OllisGit/OctoPrint-SpoolManager/pull/95) and will be in one of the next releases.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

OllisGit, 2021-01-24:
Hi @xrxca,

fyi: the qr-code feature is in lates release 1.3.0

closed 2021-05-24 by OllisGit