OllisGit/OctoPrint-SpoolManager #83: [Feature Request] Spool Length Remaining in Sidebar

charredchar, 2020-11-11:
I keep SpoolManager Plugin on my Sidebar for a quick look at how much is left on my current spool to see if it fits a job or if I need to switch it, say a large item needs printing. Issue is everywhere in OctoPrint it shows required filament as length and not weight, while SpoolManager does show remaining length if you drill down to the spool details it would be nice if it was added to the sidebar as well for a much quicker idea.

Example:
Current: "PLA-White (543.0g / 54.3%)"
Wanted: "PLA-White (543.0g / 54.3% / 176.8m)"

OllisGit, 2021-01-17:
Hi @charredchar,

during development I added this info the the sidebar, but then the Spool-Label was to long in the default view settings.

But I am thinking about an option to user-customize the label output. Somethink like this
```
[material]-[colorName] ([totalWeight] )
```
In your case
```
[material]-[colorName] ([totalWeight] / [weightProcentage] / [usedLength])

```
Which this approach each user is self responsible for the output.

What do you think?

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

charredchar, 2021-02-03:
Sorry for the late response for your question, too! heh I think that is a splendid idea! Personally I don't even really need percentage there as the two biggest uses for me is weight and length as that is what slicers and OctoPrint show as requirements for prints. I eagerly await the update.

closed 2022-01-25 by github-actions[bot]

MrBullet, 2024-12-21:
i needed exactly same thing and i made it. You can try this. I already opened pull-request so you can check what codes changed from original plugin.
https://github.com/MrBullet/OctoPrint-SpoolManager