OllisGit/OctoPrint-SpoolManager #84: Nearly new spool, warning of not enough filament.

mikelane, 2020-11-11:
It says there is -320.6995419...etc grams remaining.

[image]

AtomiKillswitch, 2020-12-17:
Check the density value for this spool.  There may be a typo here for your spool density.  If so, both the amount of filament used so far as well as the filament needed for the next print would be greatly overestimated.  See the reference the plugin suggests for density:
https://bitfab.io/blog/3d-printing-materials-densities/

This happened to me after importing a csv that had the columns in the wrong order - my total spool weight got imported into density so first few prints gave nonsense usage numbers until I fixed the density.

mikelane, 2020-12-19:
I double checked the value, it is correct as far as I can tell.

jrobbom5, 2020-12-25:
I'm having a very similar issue. This just started for me about a week ago, everything was working fine before that. Now, before every print, and for every spool, I get a warning saying "Insufficient Filament". I have double checked that the density is correct. This error happens even on spools that have been in my list for months, and have previously worked fine, but now they through this warning.

OllisGit, 2021-01-17:
Hi,
could you please provide a screenshot of the selected spool which raise the popup.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

jrobbom5, 2021-01-17:
Hi Ollis. Thanks for the reply. Hope that you had a nice break :) I'm sorry to say that it was completely my fault, nothing at all to do with Spool Manager. I had installed Filament Manager to see if it was easier to use, and forgot to uninstall it. It was Filament Manager that was throwing the error. I uninstalled it and it is now fixed. 

On a different note, I do have a usability request though that I hope that you can consider. Sometimes, I forget to update which spool I am using in Spool Manager, and that means that the weights of the spools get out of sync, so I need to reweigh the affected spools update the weights in spool manager. It would be good if I could just enter in the current weight of the spool, as spool manager already knows the original weight and the weight of the spool, so it is very easy to work out therefore how much filament has been used. Currently, I have to do this manually, ie I weigh the spool to get the current wright, then I have to look up the original weight (say 1000g) and the weight of the spool (say 134g), add them together (so 1134g), and subtract the current weight to find the used weight. This is easy math, but it is many more steps that should be required. I should just be able to enter the current weight and press "Save". Anyway, just a suggestion, and I hope that you give it consideration. 

In any case, thanks for making this plugin, it is one of my favorite Octo plugins :)

OllisGit, 2021-01-24:
Hi @jrobbom5,
thx for the feedback.

Regarding your request, other people (and myself) are thinking about a good solution about the "correct" auto-calculation.
See labeled issues: https://github.com/OllisGit/OctoPrint-SpoolManager/labels/FEATURE%3A%20WEIGHT%2FLENGH%20CALCULATION
 
Please continue your discussion in one of the issues, this one is closed now.

closed 2021-01-24 by OllisGit